cesanta/mongoose #745: Missing Unit Test Suite

jonahharris, 2016-11-23:
First, I'd like to say thank you for committing your amalgamation tools; they are *extremely* helpful! Honestly, you should publish those as separate tools altogether. As a longtime C developer, I've used and written several amalgamation tools, but none as good as yours.

Second, on the topic of this issue, would you be able to commit your Mongoose unit tests? At present, I'm testing using a modified version of the old Fossa unit test suite, which works but isn't consistent with what you have internally.

Thanks again!

mmikulicic, 2016-11-24:
would it be acceptable for you if you had to clone a submodule containing the test utils ?

osDanielLee, 2016-11-28:
@jonahharris I totally agree with you about add unit test suite thing, even it isn't useful for user, but here, all the people wants to commit code for this project would be happy to see that.

jonahharris, 2017-09-07:
@mmikulicic any way we to get a copy of the unit test suite?

closed 2018-02-02 by dimonomid

jonahharris, 2018-02-02:
You guys rock. Thanks!

dimonomid, 2018-02-02:
Actually, even though I added tests to the public repo, I realized they don't work from the public repo due to a few differences between public and private repo. :) Hopefully will fix it soon.

dimonomid, 2018-02-05:
Now, finally, tests work from the public repo. :)