cesanta/mongoose #746: problem about mg_mgr_poll

xiaxiaoyu1988, 2016-11-24:
I add a count code in the mongoose.c function mg_mgr_poll.
`
int count=0;
time_t mg_mgr_poll(struct mg_mgr *mgr, int timeout_ms) {...
  for (nc = mgr->active_connections; nc != NULL; nc = tmp) {
    int fd_flags = 0;
    if (nc->sock != INVALID_SOCKET) {
      if (num_ev > 0) {
        fd_flags = (FD_ISSET(nc->sock, &read_set) &&
                            (!(nc->flags & MG_F_UDP) || nc->listener == NULL)
                        ? _MG_F_FD_CAN_READ
                        : 0) |
                   (FD_ISSET(nc->sock, &write_set) ? _MG_F_FD_CAN_WRITE : 0) |
                   (FD_ISSET(nc->sock, &err_set) ? _MG_F_FD_ERROR : 0);
        if(fd_flags &_MG_F_FD_CAN_READ)
        {
                my_cout++;
        }
      }
#ifdef MG_LWIP
      /* With LWIP socket emulation layer, we don't get write events */
      fd_flags |= _MG_F_FD_CAN_WRITE;
#endif
    }
    tmp = nc->next;
    mg_mgr_handle_conn(nc, fd_flags, now);
  }
DBG(("=====active connections:%d", ,count));
...
}`
The http packet send program sent 2000 http reqeustes.
Normally, the count's value will be 4000. But the value is still 3005, i test a lot of times.

cpq, 2016-11-24:
Could you structure your question please?
I'm closing this.

closed 2016-11-24 by cpq

xiaxiaoyu1988, 2016-11-25:
Ok, I've found the reason. Due to select in Linux under the default support for 1024 fds.