cesanta/mongoose #748: Question: Expected granularity of mg_set_timer()

nkolban, 2016-11-24:
I am reading with great interest the mg_set_timer() API docs and how it takes a "double" (seconds) specifying the interval.  I can see lots of applicability for this in projects.  However I have questions about the possible levels of granularity.  In FreeRTOS, we have the concept of a tick, which I believe to be about 10ms.  I need a granularity that would ideally be at the  1ms level.  Do you have any thoughts (given your knowledge of internals) of what one might hope for.  For example, would mg_set_timer(0.001) be useful for a 1 msec interval?  While I realize that the timer is allowed to expire at the interval OR later ... I'm hoping that on average, if we are otherwise idle, I'm wondering what the granularity of the timer might be?

The back story is that I plan on using this on an ESP32 platform.

rojer, 2016-11-27:
mg_timers are not intended to be high precision. they are intended to be used for less critical things - think connection retries, maybe taking measurements once every few seconds, things like that.
handling of mg_timers (and miot_timers, which are implemented using mg_timers) happens during mongoose poll. on ESP8266 mongoose event loop is driven by system timers, and if there's nothing else going on, i'd expect timing to be fairly precise even down to microsecond level, however the "if nothihg else is going on" is an important caveat - anything else happening on any of the connections managed by mongoose can delay your timer. for longer delays it often doesn't matter (60000 or 60000.123 - who cares), but for millisecond intervals you are bound to see significant jitter which would likely be unacceptable.

rojer, 2016-11-27:
for single-millisecond and higher precision i'd recommend a hardware timer and an ISR.

when porting to ESP32 mg_timers, will most likely have resolution of freertos system tick, which is 10ms as you pointed out.

nkolban, 2016-11-27:
Thank you sir.

closed 2016-11-27 by nkolban