cesanta/mongoose #749: Add a method to modify connect default time in non-blocking mode

osDanielLee, 2016-11-28:
When I used mongoose in my project, I found a client try to connect a server with a wrong ip didn't exist, the timeout returned need a minute at least. Then I learnt the source code and know the reason. Due to control the timeout by user setting, I add some code in mongoose.c and mongoose.h,  and pull a request but no one handle. So I create a issue here for knowing if anyone else also need this feature, and hope the members of mongoose project to take a note for it.

Any question including assisting test is welcome and feel free to talk me that.

psykokwak-com, 2020-04-04:
Seems this PR does not work.
Does any one has a solution for that ?

psykokwak-com, 2020-04-09:
Ok. I "fixed" this issue using a timer : 

```
void TCVideoClientStonkam::mgHandler(struct mg_connection *nc, int ev, void *evdata)
{
  switch (ev) {
  case MG_EV_CONNECT:
    SDL_Log("TCVideoClientStonkam::MG_EV_CONNECT");
    break;

  case MG_EV_CLOSE:
    _connectionWasClosed = true;
    break;

  case MG_EV_TIMER:
    if (nc->flags & MG_F_CONNECTING)
      nc->flags |= MG_F_CLOSE_IMMEDIATELY;
    break;
  }
}
```

```
    mg_connection *nc = mg_connect_ws_opt(&mgr, _mghandlerfn, opts, url.c_str(), "MyApp", NULL);
    if (!nc)
      continue;

    mg_set_timer(nc, 1000));

    _connectionWasClosed = false;
    while (_networkThreadRunning && !_connectionWasClosed)
    {
      mg_mgr_poll(&mgr, 100);
    }
```

cpq, 2020-12-13:
Closing this.
Using timers is a correct way to handle the issue.

closed 2020-12-13 by cpq