cesanta/mongoose #751: Does mongoose close the websocket when ping frame timeout?

wtiandong, 2016-11-30:
Hi,
Does mongoose close the websocket when ping frame timeout? I notice mongoose sends ping frame to an opened websocket client every 5s, but doesn't check ping frame timeout? My client doesn't reply pong to mongoose server, and the server does nothing with it?

BR,
Tiandong

mmikulicic, 2016-11-30:
The main use case for ws heartbeat control frames is to keep the connection up when there is no traffic even in case of nat boxes, load balancers, proxies etc; 

That said, even without checking timeouts at the application level, if your client doesn't respond because is dead, the underlying TCP connection should be closed if it doesn't acknowledge incoming data for long enough. Without heartbeats there would be no way to tell whether the TCP connection is actually dead if an endpoint simply vanishes without sending a FIN packet.

The spec says that ping frames should be responded to though.

wtiandong, 2016-12-01:
hi mmikulicic,
But even I pull out the network wire, mongoose does nothing either ... My client connects to mongoose, and I pull out the the network wire of mongoose side, mongoose does nothing ...

alashkin, 2016-12-06:
@wtiandong 

Does it looks like https://github.com/cesanta/mongoose/issues/706?

wtiandong, 2016-12-09:
@alashkin Yes it looks like #706， but I run on Mac ... anyway, I think there should be some timeout mechanism (like MG_WEBSOCKET_PING_TIMEOUT event) which can provide a way for detecting disconnection.

alashkin, 2017-03-23:
Unfortunately, this is how sockets work - they are buffered and disconnection event may not be received.  
But on application level, you can catch WEBSOCKET_OP_PONG and drop connection is it is missed.
It can be easily done using MG_EV_TIMER/mg_set_timer.
Not sure if it should be in mongoose itself

closed 2017-03-23 by alashkin