cesanta/mongoose #752: Out of bounds memory access

mredd, 2016-12-01:
Most of the time (probably every time) a given string is compared against a fixed string, `memcmp` is used. A typical line looks like:
```c
if(memcmp(str, "ssl://", 6) == 0)
```

This is an out of bounds access, if `strlen(str) < 5` is valid, i.e. when setting the port to the typical port `"80"`. While this out of bounds read access is in most of the cases uncritical, it fails under (at least) these two circumstances:
1. When `str` is allocated close to a page boundary and the out of bounds read walks across the page boundary into the next page, the kernel throws an error if the page is not allocated by the process.
2. When using an address checker like valgrind or clang's or gcc's builtin address sanitizer, this construct triggers the address sanitizer to stop the application. That's what these tools are built for: detecting out of bounds memory access.

The proposed fix is to use `strcmp(str, "ssl://")` or `strncmp(str, "ssl://", 6)` in these cases.

mredd, 2016-12-02:
I provided a possible fix in pull request #754. Please review and comment.

alashkin, 2016-12-06:
Acrually, it is fixed in https://github.com/cesanta/mongoose/commit/fba32acb73a6b2ee9453152db08b5046e6fc539e

Sorry, I forgot to update an issue

closed 2016-12-06 by alashkin