cesanta/mongoose #753: Using Mongoose to send PUT, DELETE and other HTTP methods ...

nkolban, 2016-12-01:
I was reviewing the APIs to send outbound REST requests and don't seem to see that I can send a request with an arbitrary HTTP protocol method.  Is this correct?  I may need to send PUT and DELETE requests as well as the possibility of sending POST with no data.

alashkin, 2016-12-02:
Using `mg_connect_http_opt` - yes
But you can use `mg_connect_opt` (i.e. not _http_opt), then call `mg_set_protocol_http_websocket` and then use mg_printf & Co to send whatever you want.
But yes, unfortunately there isn't API like `mg_send_put`

closed 2016-12-02 by nkolban

Bklyn, 2018-07-19:
All I need is support for ``PUT``

What you've suggested as a solution is a pretty large amount of code for something that seems like it should be very simple.  In my code I can either:

1. Expose the private function ``mg_connect_http_base`` and reproduce the rest of ``mg_connect_http_opt`` verbatim, which is about 25-30 LOC
2. Copy the logic of ``mg_connect_http_base`` and then call ``mg_connect_opt`` and then reproduce the rest of ``mg_connect_http_opt``, so probably 50-60 LOC.

As a user of this wonderful library I'd much rather have a variant of ``mg_connect_http`` or ``mg_connect_http_opt`` that lets me pass the method explicitly.