cesanta/mongoose #755: Handling reconnection attempts after connection failure

readysteadygo2006, 2016-12-03:
Hi,

I'm using mongoose's websocket client functionality.

I'm trying to initiate a reconnection attempt in the even of a failed connection.

My ev_handler code is taken from the chat client with a simple extension to reinitialise the connection
and mgr and free to reconnect.

 static void ev_handler_clnt(struct mg_connection *nc, int ev, void *ev_data) {
  struct websocket_message *wm = (struct websocket_message *) ev_data;
  (void) nc;

  printf("handler call nc: %p\n", nc);
  switch (ev) {
    case MG_EV_CONNECT: {
      printf("-- MG_EV_CONNECT\n");
      int status = *((int *) ev_data);
      if (status != 0) {
        printf("-- Connection error: %d:%s\n", status, strerror(status));
        vTaskDelay(2000 / portTICK_RATE_MS);
        printf("-- Trying to reconnect\n");
        skip_close_client = 1;
//        mg_mgr_free(&mgr);
        struct mg_connection *recon_nc;
        mg_mgr_init(&mgr, NULL);
        recon_nc = mg_connect_ws(&mgr, ev_handler_clnt, chat_server_url, NULL, NULL);
        printf("recon_nc: %p\n", recon_nc);
        if (recon_nc == NULL) {
            printf("Invalid address\n");
            s_done_client = 1;
        }
      }
      break;

If I dont free mgr, I get reconnection attempts but eventually run out of memory. If I free mgr
I just get  MG_EV_CLOSE events.

Can you help?

BR
Paul Grant

osDanielLee, 2016-12-08:
I suggest you don't do the reconnection in ev_handler_clnt func. As you said, it'll cause some problems. My solution for reconnection is that judge a flag modified in ev_handler_clnt (represent the connect work well or not, just like s_exit_flag in example https://github.com/cesanta/mongoose/blob/master/examples/restful_client/restful_client.c, but need to return another value like -1 for MG_EV_CLOSE), if the flag != 1 (connect didn't be built), ignore this connection and try it again in the main func (still take a example as https://github.com/cesanta/mongoose/blob/master/examples/restful_client/restful_client.c). More about that, you could add a counter for reconnection, it'll look like more sensible.

cpq, 2017-02-03:
@readysteadygo2006 , the manager should never be freed during the lifetime of the application.

closed 2017-02-03 by cpq