cesanta/mongoose #756: user_data assigned to an mg_mgr not available through a new connection in multithreaded mode

dsmccall, 2016-12-08:
I tried supplying a user_data pointer during mg_mgr_init - I was intending to then retrieve this from the `mg_connection::mgr` pointer in my event handler.  It works perfectly in single-threaded mode, but the pointer is NULL in multi-threaded mode.  Here's an example of how to reproduce it (a simple addition to the multithreaded_restful_server.c example).  This is using version 6.6 (master branch downloaded on 2016/12/07).

#include "mongoose.h"

static const char *s_http_port = "8000";

static void ev_handler(struct mg_connection *c, int ev, void *p) {
    if (ev == MG_EV_HTTP_REQUEST) {
        struct http_message *hm = (struct http_message *) p;
        char reply[100];

        // Get user data from connection manager
        int* simple_user_data = (int*)c->mgr->user_data;

        /* Send the reply */
        snprintf(reply, sizeof(reply), "{ \"simple user data\": \"%i\" }\n", *simple_user_data);
        mg_printf(c,
            "HTTP/1.1 200 OK\r\n"
            "Content-Type: application/json\r\n"
            "Content-Length: %d\r\n"
            "\r\n"
            "%s",
            (int)strlen(reply), reply);
    }
}

int main(void) {
    struct mg_mgr mgr;
    struct mg_connection *nc;

    int simple_user_data = 12345;

    mg_mgr_init(&mgr, (void*)&simple_user_data);
    nc = mg_bind(&mgr, s_http_port, ev_handler);
    mg_set_protocol_http_websocket(nc);

    /* For each new connection, execute ev_handler in a separate thread */
    //mg_enable_multithreading(nc);

    printf("Starting multi-threaded server on port %s\n", s_http_port);
    for (;;) {
        mg_mgr_poll(&mgr, 3000);
    }
    mg_mgr_free(&mgr);

    return 0;
}

mat007, 2016-12-09:
Hi,

I don't know whether this issue is a bug or not, however you can work around it by using the connection user data, e.g. binding like this:

    mg_bind_opts opts;
    memset( &opts, 0, sizeof( opts ) );
    opts.user_data = &simple_user_data;
    nc = mg_bind_opt( &mgr, s_http_port, ev_handler, opts );

Then in the event handler you simply retrieve the user data like this:

    int* simple_user_data = (int*)c->user_data;

I hope this helps.

dsmccall, 2016-12-09:
It does indeed help, and I can also work around it by attaching directly to the user_data pointer of the new connection after the call.

    nc->user_data = (void*)this;

You may be right that this isn't a bug - however, it seems inconsistent that it works in single-threaded mode but not in multi-threaded.  That may well be the intended operation!

Thank you for your help, very much appreciated.

cpq, 2017-04-14:
Closing, as multithreaded mode is no more.

closed 2017-04-14 by cpq