cesanta/mongoose #758: About multi-threading

mat007, 2016-12-09:
Hi,

I am currently upgrading from mongoose 3.3 to 6.6 a server which receives a fair amount of concurrent requests, some of which can take some time and thus require some kind of parallelism.
My first attempt was to mg_enable_multithreading as being done in the multithreaded_restful_server example, however this fails a bit short for a couple of reasons:
. there is no means to define a pool of threads and it is just not practical to start one thread per connection in real world servers
. there appears to be no join or anything to correctly stop a server without leaving dangling event handlers calling into already destroyed data

Now I'm thinking to leave mongoose single threaded and handle a pool of worker tasks started upon receiving MG_EV_HTTP_REQUEST events.
This will also require to manage connections lifetime separately, mainly to listen to MG_EV_CLOSE events in order to invalidate connections and abort the matching task.
In the end this might get rather cumbersome and pushes a lot of complexity to the user...

Did I miss anything?

Thank you!

FraMecca, 2017-01-09:
Same problem here. it is inconvenient to have a thread per connection when parallelism is needed only in a subset of the tasks.

KingDuckZ, 2017-01-13:
Same. I also got to the same conclusion as mat007, and I'm also not convinced for the same reasons he listed.

I found [this discussion](https://groups.google.com/forum/#!topic/mongoose-users/i4TTlT_uMSk), where it's suggested to (copy paste):

1. create first manager, and bind it to the port:
```cpp
    struct mg_connection *c = mg_bind(mgr1, s_http_port, ev_handler);
```
2. from the listening connection, get the socket:
```cpp
    sock_t sock = c->sock;
```
3. Start other managers in the loop, and add listening socket to them by cloning the first one:
```cpp
    //....
    struct mg_connection *c = mg_add_sock(mgrN, dup(sock), ev_handler);
```

I did as suggested, but all the sockets in my worker threads get closed immediately on the first request and then stay dead. By adding <code>fprintf(stderr, "error %d: %s\n", errno, strerror(errno));</code> in mongoose.c:3426 I could figure out that recv (or whatever is behind MG_RECV_FUNC) returns -1 and sets errno to ENOTCONN (107, "Transport endpoint is not connected"). As a consequence of the error MG_F_CLOSE_IMMEDIATELY gets set to the corresponding sock's flags and my event handler only ever receives one notification (aside of the initial POLL notifications). From there on, all but the main manager are completely dead. Am I using <code>dup()</code> the wrong way, is that post obsolete/wrong...?

f4bsch, 2017-02-10:
Its not too hard to get mongoose multithreaded, see my implementation (draft) at https://github.com/rtaudio/masaj/blob/master/JsonHttpServer.cpp .

It creates a threadpool when the server starts up and processes 2 mongoose events: 

**MG_EV_HTTP_REQUEST**: Here we decide wether to process this request in a worker thread or in the main thread.
For static files, `mg_serve_http` should always be processed in the main thread, since it uses buffers and they are not thread safe. It wont send files at once, so it should not be a problem with blocking.
For async request handling (such as streams), I push a task into the thread pool with 2 parameters (besides general binding and HTTP request message):
* handle to mg_connection
* a condition variable to notify the main thread that our worker thread is done

A handle to the condition variable is stored in `nc->user_data`, and later used for MG_EV_CLOSE events.
Then in the worker thread you can directly write to `conn->sock`, `mg_send` cannot be used since it uses interal buffers that are not thread safe.
When the worker thread is done, it should set `nc->flags |= MG_F_CLOSE_IMMEDIATELY;`, and also send "Connection: close" HTTP header.

**MG_EV_CLOSE**: Here you just check if the connection has been handled async, set a closed flag to let the worker thread know that it should shut down, then wait for the condition variable to block until the worker thread ends.

A drawback of this is that you have to work on raw sockets, it might be more elegant to create a new manager for each socket like @KingDuckZ said .

stromnet, 2017-02-10:
Writing to socket from another thread seems, as you say, not an ideal solution.
I have a similar jsonrpc server (not commited anywhere yet due to #765). I'm doing lenghty stuff in background thread, storing the result in memory, then waking up the poll loop using mg_broadcast and doing the actual socket write in the MG_EV_POLL callback. Never touching mg_conn* from outside of mgr_poll callbacks.

cpq, 2017-02-11:
the way @f4bsch  describes should work, although it's a bit clunky to work with a non-blocking socket.

here's another idea, although i might miss some details. all IO goes via mbufs. thus, if we mutex-protect mbufs, we should be set. It should be OK to give `struct mg_connection *` pointer to another thread - the only thing to care about is to get notified about close events , when pointer gets invalidated. That could be done as @f4bsch says, via the condvar.

cpq, 2017-02-11:
@stromnet actually if the data that is sent out is not large, you can send it directly by `mg_broadcast`, and not bother with `MG_EV_POLL`.

The @stromnet is the "recommended", anticipated way of doing multithreading with mongoose.

stromnet, 2017-02-11:
@cpq: sending data directly with mg_broadcast only works if the data should go to all the connected sockets though, right? Since mg_broadcast handler would be called on all sockets with this data.

cpq, 2017-02-13:
@stromnet that's correct, `mg_broadcast()` will make the callback to be called on each connection.
You can, however, pass an identifier of the target connection as part of your data.
Then, a callback function can send the data to only that connection.

cpq, 2017-03-15:
Closing this.

closed 2017-03-15 by cpq