cesanta/mongoose #765: segfault comes occasionally by using CGI

guangjung, 2016-12-19:
We debuged code based the lastest code . We found  mongoose server called nc->iface->vtable->tcp_send may be segfault occasionally by using CGI.  And I found that the pointer  nc->iface->vtable->tcp_send  was not NULL. 

[ 2264.502148] mongoose[2965]: segfault at 8 ip 000000000804e12e sp 00000000ff9e9550 error 4 in mongoose[8048000+19e000] 

void mg_send(struct mg_connection *nc, const void *buf, int len) {
  nc->last_io_time = (time_t) mg_time();
  if (nc->flags & MG_F_UDP) {
    nc->iface->vtable->udp_send(nc, buf, len);
  } else {
    nc->iface->vtable->tcp_send(nc, buf, len);
  }
#if !defined(NO_LIBC) && MG_ENABLE_HEXDUMP
  if (nc->mgr && nc->mgr->hexdump_file != NULL) {
    mg_hexdump_connection(nc, nc->mgr->hexdump_file, buf, len, MG_EV_SEND);
  }
#endif
}

cpq, 2017-02-03:
Could you share more details about your environment please - OS, CGI script used, etc?

stromnet, 2017-02-04:
I think I have a similar crash. I can reproduce it by calling using curl towards mongoose, which executes a CGI (no handling of that request on our end besides forwarding it to mg_serve_http). 
If I SIGINT the curl process before the response has been sent by the server, it segfaults on the above location:

```
(gdb)
#0  0x00000000004662a5 in mg_send (nc=0x80a82e2c0, buf=0x80a833c00, len=1024) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:2515
#1  0x0000000000468a8d in mg_forward (from=0x80a82e3a0, to=0x80a82e2c0) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:2956
#2  0x000000000047add3 in mg_cgi_ev_handler (cgi_nc=0x80a82e3a0, ev=3, ev_data=0x7fffdfdfc93c) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:8267
#3  0x00000000004659b0 in mg_call (nc=0x80a82e3a0, ev_handler=0x47ab60 <mg_cgi_ev_handler>, ev=3, ev_data=0x7fffdfdfc93c) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:2106
#4  0x0000000000466d95 in mg_recv_common (nc=0x80a82e3a0, buf=0x80a833c00, len=1024, own=1) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:2557
#5  0x0000000000466be9 in mg_if_recv_tcp_cb (nc=0x80a82e3a0, buf=0x80a833c00, len=1024, own=1) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:2561
#6  0x0000000000469b66 in mg_handle_tcp_read (conn=0x80a82e3a0) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:3424
#7  0x0000000000469718 in mg_mgr_handle_conn (nc=0x80a82e3a0, fd_flags=1, now=1486217627.5229161) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:3549
#8  0x000000000046aa6b in mg_socket_if_poll (iface=0x80845fa80, timeout_ms=1000) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:3742
#9  0x0000000000466080 in mg_mgr_poll (m=0x7fffffffe8f0, timeout_ms=1000) at /usr/home/johan/dev/agocontrol/core/rpc/mongoose.c:2287
(gdb) print *nc
$1 = {next = 0x0, prev = 0x0, listener = 0x0, mgr = 0x0, sock = 0, err = 0, sa = {sa = {sa_len = 0 '\0', sa_family = 0 '\0', sa_data = '\0' <repeats 13 times>}, sin = {sin_len = 0 '\0', sin_family = 0 '\0', sin_port = 0, sin_addr = {
        s_addr = 0}, sin_zero = "\000\000\000\000\000\000\000"}, sin6 = {sa_len = 0 '\0', sa_family = 0 '\0', sa_data = '\0' <repeats 13 times>}}, recv_mbuf_limit = 0, recv_mbuf = {buf = 0x0, len = 0, size = 0}, send_mbuf = {buf = 0x0,
    len = 0, size = 0}, last_io_time = 1486217627, ev_timer_time = 0, proto_handler = 0, proto_data = 0x0, proto_data_destructor = 0, handler = 0, user_data = 0x0, priv_1 = {v = 0x0, f = 0}, priv_2 = 0x0, mgr_data = 0x0, iface = 0x0,
  flags = 0}
```

My guess is that the frontend connection has been closed, but the backend (CGI stdout/err pipe) forwarding does not detect this and tries to write anyway?

stromnet, 2017-02-04:
Perhaps unrelated, perhaps not. Saw this when running with valgrind and closing the connection, perhaps before it had time to send, not sure (I patched mg_send to return if nc->iface was null)
```
==59392== Thread 3:
==59392== Invalid read of size 8
==59392==    at 0x47B235: mg_cgi_ev_handler (mongoose.c:8245)
==59392==    by 0x465FEF: mg_call (mongoose.c:2106)
==59392==    by 0x4673F4: mg_recv_common (mongoose.c:2558)
==59392==    by 0x467248: mg_if_recv_tcp_cb (mongoose.c:2562)
==59392==    by 0x46A1C5: mg_handle_tcp_read (mongoose.c:3425)
==59392==    by 0x469D77: mg_mgr_handle_conn (mongoose.c:3550)
==59392==    by 0x46B0CA: mg_socket_if_poll (mongoose.c:3743)
==59392==    by 0x4666BF: mg_mgr_poll (mongoose.c:2287)
...

==59392==  Address 0xef8ab68 is 200 bytes inside a block of size 208 free'd
==59392==    at 0x4C232CC: free (in /usr/local/lib/valgrind/vgpreload_memcheck-amd64-freebsd.so)
==59392==    by 0x468141: mg_destroy_conn (mongoose.c:2156)
==59392==    by 0x4667A0: mg_close_conn (mongoose.c:2164)
==59392==    by 0x46B150: mg_socket_if_poll (mongoose.c:3750)
==59392==    by 0x4666BF: mg_mgr_poll (mongoose.c:2287)
...
```

f4bsch, 2017-02-09:
I might have encountered a related issue on Windows 10 and Ubuntu 16.

On Windows 10:
There is an occasional heap corruption, that is detected at `mg_destroy_conn() / MG_FREE(conn);`
I noticed that `conn->iface` and `conn->flags` have garbage values, sometimes even on a first `ev_handler` call for an incoming HTTP request.
Sometimes it crashes at `select()` in `mg_socket_if_poll`.

On Ubuntu 16:
In `mg_http_conn_destructor` crash at `free(proto_data)` (double free or corruption)
Stack:
```
0	__GI_raise	/usr/lib/debug/lib/x86_64-linux-gnu/libc-2.23.so	54	0x7ffff6c85428	
1	__GI_abort	/usr/lib/debug/lib/x86_64-linux-gnu/libc-2.23.so	89	0x7ffff6c8702a	
2	__libc_message	/usr/lib/debug/lib/x86_64-linux-gnu/libc-2.23.so	175	0x7ffff6cc77ea	
3	malloc_printerr	/usr/lib/debug/lib/x86_64-linux-gnu/libc-2.23.so	5004	0x7ffff6ccfe0a	
4	_int_free	/usr/lib/debug/lib/x86_64-linux-gnu/libc-2.23.so	3865	0x7ffff6ccfe0a	
5	__GI___libc_free	/usr/lib/debug/lib/x86_64-linux-gnu/libc-2.23.so	2966	0x7ffff6cd398c	
6	mg_http_conn_destructor	mongoose.c	4970	0x4b15dd	
7	mg_destroy_conn	mongoose.c	2091	0x4ac90e	
8	mg_close_conn	mongoose.c	2108	0x4ac9f0	
9	mg_socket_if_poll	mongoose.c	3693	0x4b04ff	
10	mg_mgr_poll	mongoose.c	2231	0x4acdf0	
...	<More>				
```

stromnet, 2017-02-09:
I've seen other unexplainable wierdnesses with CGI, but haven't been able to pinpoint it.. Some  heap corruption could very well be the reason.

The symptoms is that std::clog gets it's failbit (or if it was badbit) set, and my logging stops working. In some cases the app hangs too, but after I enabled exception-throwing in case of badbit/failbit, it always stops at that point (app crashes/breaks into debugger instead of continuing and further breaking apart).

I can reproduce this by calling an URL with CGI endpoint repeatedly, sooner or later it happens.
Requests to non-existing CGI files are not a problem, neither are other static resources or other resources handled by my app.

This is on FreeBSD, haven't tested the latest commits yet though.

stromnet, 2017-02-09:
Tested master, same problem.

I've been able to reproduce it easier with `valgrind --free-fill=0xda` which gave me a similar trace to the one I pasted before:

```
==77195== Thread 3:
==77195== Invalid read of size 8
==77195==    at 0x47B5B5: mg_cgi_ev_handler (mongoose.c:8070)
==77195==    by 0x46620F: mg_call (mongoose.c:2050)
==77195==    by 0x467614: mg_recv_common (mongoose.c:2502)
==77195==    by 0x467468: mg_if_recv_tcp_cb (mongoose.c:2506)
==77195==    by 0x46A3E5: mg_handle_tcp_read (mongoose.c:3369)
==77195==    by 0x469F97: mg_mgr_handle_conn (mongoose.c:3494)
==77195==    by 0x46B2EA: mg_socket_if_poll (mongoose.c:3687)
==77195==    by 0x4668DF: mg_mgr_poll (mongoose.c:2231)
...

==77195==  Address 0xf408ad8 is 200 bytes inside a block of size 208 free'd
==77195==    at 0x4C232CC: free (in /usr/local/lib/valgrind/vgpreload_memcheck-amd64-freebsd.so)
==77195==    by 0x468361: mg_destroy_conn (mongoose.c:2100)
==77195==    by 0x4669C0: mg_close_conn (mongoose.c:2108)
==77195==    by 0x46B370: mg_socket_if_poll (mongoose.c:3694)
==77195==    by 0x4668DF: mg_mgr_poll (mongoose.c:2231)
...
```
 This made me dig a bit, and I think this may be a problem:

1. mg_handle_cgi creates new socket cgi_nc
1. cgi_nc gets it's proto_data inited:
```
    struct mg_http_proto_data *cgi_pd = mg_http_get_proto_data(cgi_nc);
    cgi_pd->cgi.cgi_nc = cgi_nc;
    cgi_pd->cgi.cgi_nc->user_data = nc;
```
 *BUT*: `nc` gets no reference to cgi_nc as far as I can tell
1. at some point, nc is marked for close, and is closed from `mg_socket_if_poll` cleanup part (which checks for  immediate close or send_and_close). Perhaps due to premature close (before cgi responds)?
1. eventually mg_cgi_ev_handler is called for the backend socket, which obtains pointer to the *now free'd* nc through it's user_data:
```
struct mg_connection *nc = (struct mg_connection *) cgi_nc->user_data;
```
1. then it checks nc->flags, which is a use after free, and in some cases writes to flags.

So, if my theory is correct the frontend connection needs to know about the backend connection, so it can clear the backends user_data.

f4bsch, 2017-02-09:
I just fixed my issue using AddressSanitizer and tracked it down to size difference of the mg_connection struct between C and C++. This was caused by an inproper definition of `MG_ENABLE_IPV6` so `union socket_address` had a different size in included header and when compiling `mongoose.c`.

@stromnet Make sure that all `MG_*` feature macros are properly defined by your build tools.

I used UndefinedBehaviorSanitizer btw, and these flags for gcc: `-fsanitize=leak -fsanitize=undefined -fsanitize=bounds -O1 -fsanitize=address -fno-omit-frame-pointer -fno-optimize-sibling-calls`

stromnet, 2017-02-10:
Hm, I do not have any custom MG_ defs set either on mongoos.c or on my app. And MG_ENABLE_IPV6 seems consistently set in mongoos.c/h (.h sets it to 0 unless defined).

Just to understand, in what way does your code use the `struct mg_connection` so that something like that would make a difference? I'm using quite a small basic subset of the library's functionality, but I only ever deal with pointers to mg_connections, so shouldn't matter anyway.

Regarding the sanitize flags, CLANG seems to have similar, but not on FreeBSD :/ At least not without some extra effort (installing custom version etc). May have to look into that though, thanks for the tip!

f4bsch, 2017-02-10:
I process some requests asynchronously, each worker bound to the `mg_connection` pointer and when the request handler is done its sets `nc->flags` to MG_F_CLOSE_IMMEDIATELY to let the mgr know that it should close the connection. Thats where in my case the heap got corrupted, because `nc->flags` pointed to out of bound address.
Do you run your CGI in the ev_handler or in another thread?

stromnet, 2017-02-10:
Ah, okey, that makes sense.

I have one thread which only does `mg_mgr_poll(&mongoose_mgr, 1000)` in a loop.
On any MG_EV_HTTP_REQUEST for unknown URLs I just call `mg_serve_http(conn, (struct http_message *) ev_data, http_serve_opts);` which will start the new process and handle everything CGI-related. Thus nothing CGI-related in my code.
All other calls to mg data are only ever made in that thread, I do have some other threads which have their own data though. When they are ready the use mg_broadcast to trigger a MG_EV_POLL, which will check for what to do through it's conn_data. The conn_data is only ever read/written by that thread, but holds an shared_ptr to app-specific data which I'm reading/writing to. When both MG and async thread are done with it it will be automatically released (but that is out of scope for this.. :)).

cpq, 2017-02-11:
@stromnet you're right, the crash is easily reproducible by simple CGI:

```
#!/bin/sh

echo Content-Type: text/plain
echo

while true; do
  date
done
```

```
curl 127.0.0.1:9001/blurp.cgi
```

When `curl` is interrupted, mongoose segfaults.
Looking into that.

cpq, 2017-02-11:
@stromnet @guangjung please verify

closed 2017-02-11 by cpq

stromnet, 2017-02-11:
@cpq: Verified, seems to work. Thanks!

stromnet, 2017-02-11:
Actually, it seems that it's not entierly good.. but much harder to reproduce.. Haven't been able to reproduce it with valgrind yet, but it does crash in a similar often-used codepath (not directly mongoose-related, it's logging called from my event handler). So something is still weird, but if it's related to to this, I cannot yet say.

stromnet, 2017-02-27:
The remaining crash was non-mongoose related, it was rather related to my SIGCHLD handling and not having SA_RESTART on signal handler. 

All of the following is totally unspecific for mongoose, but in case it bites some one else:

In short, the (boost) logging [writes to std::cerr]( https://gcc.gnu.org/onlinedocs/libstdc++/libstdc++-html-USERS-3.4/ostream-source.html#l00286). This is a `std::ostream` which writes to a `std::streambuf` implementation which is backed by the stderr file descriptor. In the std:: classes this works by calling `fwrite`() on `stderr` [1], and if the number of bytes written does not match the number of bytes it requested to write, it sets the `badbit` on the ostream. This will make std::cerr totally silent, until manually reset (`std::cerr.clear()`).

Since the first symptoms I was debugging was a suddenly silenced logging , I had set `std::cerr.exceptions ( std::ostream::failbit | std::ostream::badbit );` This means that instead of just going silent, it throws an exception, which aborts my program.

What was happening here was that the very frequent `SIGCHLD` signals interrupted the fwrite, which wrote less bytes than requested, and set errno=EINTR, causing the badbit to be set (and trigger exception).

My solution (works so far), is two part:
1. Block all signals first thing when the program starts, this is then inherited by all threads [2]
```
    sigset_t set;
    sigfillset(&set);
    pthread_sigmask(SIG_BLOCK, &set, NULL);
```

1. Register my signal handlers with SA_RESTART (which required some hack since I'm using boost::asio::signals)
```
    signals.add(signal); // signals is a boost::asio::signal_set
    struct sigaction fix;
    sigaction(signal, NULL, &fix);
    fix.sa_flags = SA_RESTART;
    sigaction(signal, &fix, NULL);
```

1. Unblock signals in one single thread which is designated to handle signals (and other async IO through a boost::asio::io_service).
```
    sigset_t set;
    sigfillset(&set);
    pthread_sigmask(SIG_UNBLOCK, &set, NULL);
```

This seems to have resolved my issues on both FreeBSD and Linux. Hope it may help someone else! :)