cesanta/mongoose #766: Event loop max CPU after running for a day or two

jmgregory, 2016-12-22:
I'm running latest Mongoose from github, with `-DMG_ENABLE_THREADS -DMG_DISABLE_HTTP_KEEP_ALIVE -DMG_ENABLE_HTTP_WEBSOCKET=0 -DMG_ENABLE_MQTT=0`.  Our server is intended to run 24/7 on an OEM appliance (ARMv7, android-18).  Also, I'm using the patch in pull request #761, so it may be that this bug only appears after that patch is applied.

We've noticed that after a day or two, the server gets into a state where it's using the full CPU, and no longer responds to HTTP requests.  Specifically, it's the event loop thread that's pegging the CPU.  In `mg_mgr_poll()` there's one `iface`, and its poll function is `mg_socket_if_poll()`.  Inside `mg_socket_if_poll()`, `iface->mgr` has 5 active connections, one of which has `flags == 1` (listening), while the other four have `flags == 0`.  All of the connections have `err == 0`.

It seems that `select()` is returning -1 and setting `errno` to 9 (bad file number), and that this is not handled anywhere.

What should be the correct behavior here?

jmgregory, 2016-12-22:
A little more info, if it helps at all:

The socket that's causing the EBADF error is on one of the connections with `flags == 0`.  Its `proto_handler` is `mg_http_handler()` and its `handler` is `forwarder_ev_handler`.

jmgregory, 2016-12-22:
Trying out this patch for now.  Obviously not the solution, but it might keep things up and running.

    @@ -3694,7 +3694,30 @@ time_t mg_socket_if_poll(struct mg_iface *iface, int timeout_ms) {
       tv.tv_usec = (timeout_ms % 1000) * 1000;
     
       num_ev = select((int) max_fd + 1, &read_set, &write_set, &err_set, &tv);
    +  int select_err = mg_get_errno();
       now = mg_time();
    +
    +  if (num_ev == -1) {
    +      LOG(LL_ERROR, ("select() error: %d (%s)", select_err, strerror(select_err)));
    +  }
    +
    +  // Check for bad file descriptors
    +  if (num_ev == -1 && select_err == EBADF) {
    +      // One or more of these sockets is no good.  Mark its connection for closing.
    +      for (nc = mgr->active_connections; nc != NULL; nc = tmp) {
    +          tmp = nc->next;
    +          if (fcntl(nc->sock, F_GETFD) == -1 && mg_get_errno() == EBADF) {
    +              DBG(("marking socket %d for closure", nc->sock));
    +              nc->flags |= MG_F_CLOSE_IMMEDIATELY;
    +          }
    +      }
    +#if MG_ENABLE_BROADCAST
    +      if (fcntl(mgr->ctl[1], F_GETFD) == -1 && mg_get_errno() == EBADF) {
    +          // Do what?
    +      }
    +#endif
    +  }
    +
     #if 0
       DBG(("select @ %ld num_ev=%d of %d, timeout=%d", (long) now, num_ev, num_fds, timeout_ms));

cpq, 2017-04-14:
@jmgregory could you confirm that the latest mongoose has this issue fixed please?

The latest code is at the dev branch, https://github.com/cesanta/mongoose/tree/dev

closed 2018-01-30 by dimonomid