cesanta/mongoose #780: Failure with ESP32 and WebSockets ...

nkolban, 2017-02-06:
I am running the latest mongoose library with the latest ESP-IDF as of the date of this post.  I have setup a Mongoose server in the ESP32 but when I try and form a connection from a browser using WebSockets, we get errors and then go into a hard loop around the call to `mg_mgr_poll()`.

Here is the core of my application ...

```
static void eventHandler(
  struct mg_connection *nc,
  int ev,
  void *evData) {
	char temp[100];
	mg_conn_addr_to_str(nc, temp, sizeof(temp),
			MG_SOCK_STRINGIFY_IP | MG_SOCK_STRINGIFY_PORT | MG_SOCK_STRINGIFY_REMOTE);
	// If the event is a poll indication, nothing further to be done.  This was just an
	// informational event.
	if (ev == MG_EV_POLL) {
		return;
	}
	ESP_LOGD(tag, "mongoose event: %d %s from %s", ev, mongoose_eventToString(ev), temp);
} // eventHandler

void task_web_socket(void *ignore) {
	ESP_LOGD(tag, ">> task_web_socket");
	struct mg_mgr mgr;
	mg_mgr_init(&mgr, NULL);
	struct mg_connection *nc = mg_bind(&mgr, ":80", eventHandler);
	if (nc == NULL) {
		ESP_LOGE(tag, "No connection from the mg_bind()");
		vTaskDelete(NULL);
		return;
	}
	mg_set_protocol_http_websocket(nc);
	while(1) {
		mg_mgr_poll(&mgr, 1000);
		ESP_LOGD(tag, "Tick");
	}
} // task_web_socket
```

The task is called when the ESP32 has been initialized and has a good IP address.  In the log from the ESP32 we see:

```
D (4799) web_sockets: >> task_web_socket
mg_socket_if_init    0x3ffcf6b0 using select()
mg_mgr_init_opt      ==================================
mg_mgr_init_opt      init mgr=0x3ffcf6b0
mg_socket_if_sock_se 0x3ffb8bec 0
mg_add_conn          0x3ffcf6b0 0x3ffb8bec
D (5824) web_sockets: Tick
D (6824) web_sockets: Tick
D (7825) web_sockets: Tick
D (8826) web_sockets: Tick
D (9827) web_sockets: Tick
D (10828) web_sockets: Tick
D (11829) web_sockets: Tick
D (12830) web_sockets: Tick
D (13831) web_sockets: Tick
mg_mgr_handle_conn   0x3ffb8bec fd=0 fd_flags=1 nc_flags=1 rmbl=0 smbl=0
mg_accept_conn       0x3ffb8bec: failed to accept: 128
mg_mgr_handle_conn   0x3ffb8bec after fd=0 nc_flags=1 rmbl=0 smbl=0
D (14616) web_sockets: Tick
mg_mgr_handle_conn   0x3ffb8bec fd=0 fd_flags=1 nc_flags=1 rmbl=0 smbl=0
mg_accept_conn       0x3ffb8bec: failed to accept: 128
mg_mgr_handle_conn   0x3ffb8bec after fd=0 nc_flags=1 rmbl=0 smbl=0
D (14636) web_sockets: Tick
mg_mgr_handle_conn   0x3ffb8bec fd=0 fd_flags=1 nc_flags=1 rmbl=0 smbl=0
mg_accept_conn       0x3ffb8bec: failed to accept: 128
mg_mgr_handle_conn   0x3ffb8bec after fd=0 nc_flags=1 rmbl=0 smbl=0
D (14656) web_sockets: Tick
mg_mgr_handle_conn   0x3ffb8bec fd=0 fd_flags=1 nc_flags=1 rmbl=0 smbl=0
mg_accept_conn       0x3ffb8bec: failed to accept: 128
mg_mgr_handle_conn   0x3ffb8bec after fd=0 nc_flags=1 rmbl=0 smbl=0
D (14676) web_sockets: Tick

```

This log shows that Mongoose starts and we are ticking away waking up from the `mg_mgr_poll` loop every second.  Once it has been ticking for a while, I run the following browser script:
```
<!DOCTYPE html>
<html>
<head>
<meta charset="UTF-8">
<title>WebSocket client</title>
<script>
// Setup a Web socket client call
console.log("Setting up web socket");
var ws = new WebSocket("ws://192.168.1.99:80");
ws.onopen = function() {
	console.log("Web Socket opened!");
}
ws.onerror = function(err) {
	console.log("Web socket error detected!");
}
ws.onclose = function() {
	console.log("Web socket closed!");
}
ws.onmessage = function(event) {
	console.log("Message received!");
}
</script>
</head>
<body>

</body>
</html>
```

which forms a web socket.  At this point, we start to see the error from "mg_accept_conn" and then we go into a tight loop in the core `while()` loop surrounding `mg_mgr_poll`.  I can recreate at will.

cpq, 2017-02-06:
Hi Neil!

Just to make sure, did you try Mongoose OS on ESP32 (https://mongoose-iot.com/) ?
Your book mentions the library, but not the OS, which has this and many other issues solved.

nkolban, 2017-02-06:
Howdy Sergey,
Hmm ... no sir, I haven't gone near the Mongoose OS yet ... my thinking was that the "Mongoose" library was a framework and processor for networking including broken out parsing and construction for HTTP, MQTT, and WebSockets ... and it would function stand-alone.   Is that not the case?

I do plan to get to evaluating Mongoose OS but my immediate plan was to focus in on "functional areas" and HTTP, MQTT and WebSockets are my foundations ...

Regarding this specific issue ... should I not consider stand-alone Mongoose as an environment for writing WebSockets apps on an ESP32?

nkolban, 2017-02-06:
Howdy Sergey,
I must say I'm a bit confused.  While I fully understand and appreciate Cesanta wanting developers and designers to use Mongoose OS ... and I have absolutely no issue with y'all promoting it ... I approached the "Mongoose - Embedded Web Server / Embedded Networking Library" ... as documented in the README here ... https://github.com/cesanta/mongoose  ... as being a self contained "Web Server and networking library" that could be leveraged as a "unit of function".  Reading the docs the features of THAT project include "WebSocket Client, WebSocket Server".   I haven't found a statement that says that in order to leverage the WebSocket Server functions, one MUST also leverage the Mongoose OS package.

Is that in fact the statement we want to end up with: "In order to utilize the Mongoose Embedded Web Server library to support WebSockets, you MUST also build upon the project known as Mongoose OS".

Obviously, all decisions y'all make are the final words ... I'm just looking to understand the story and actuality.  While I can listen to your discussion points on the value of including Mongoose OS in any projects, right now I'm concentrating on the actual mechanics and utilization realities.  If you are saying that in order to use Mongoose Web Server for Web Sockets I MUST also use Mongoose OS ... that is one story ... while if we are saying that Mongoose Web Server for Web Sockets CAN'T be used without Mongoose OS today but will be able to do so in the future ... that is a different story.

Can you clarify expectations for using Mongoose Web Server for Web Sockets on an ESP32 without also leveraging Mongoose OS?   (And a simple ... "You can't use Mongoose Web Server for Web Sockets on an ESP32 today without Mongoose OS and we don't plan to change that" would be a concluding answer).

Neil

cpq, 2017-02-06:
The library has way wider coverage than the OS. For example, you can use Mongoose Library on these: 

- [on nxp lpc](https://www.hackster.io/rojer/mongoose-embedded-networking-library-on-nxp-lpc-260945?ref=channel&ref_id=21424_trending___&offset=1)
- [on nrf 51/52](https://www.hackster.io/dfrank/mongoose-embedded-networking-library-on-nrf51-and-nrf52-547b15?ref=channel&ref_id=21424_trending___&offset=4)
- [on pic32](https://www.hackster.io/alashkin/mongoose-networking-library-on-pic32-chip-9596d5?ref=channel&ref_id=21424_trending___&offset=7)
- [on TI TIVA](https://www.hackster.io/rojer/mongoose-embedded-networking-library-on-ti-tiva-tm4c129-fb3391?ref=channel&ref_id=21424_trending___&offset=2)
- [on nxp K64](https://github.com/cesanta/mongoose/tree/master/examples/NXP_K64)

But when it comes to Espressif boards, Mongoose OS is available there, which adds much more to the table, and makes using the library a breeze, together with many extras.  So, you talk about ESP32, talk about Mongoose Library - but not a word is said about the Mongoose OS.

You don't need to use an OS. Of course you can use the library only. My point is: if you talk about ESP32 environment and touch Mongoose, it'd make sense to expand on the OS, because that's what a developer (your reader) would benefit most.

nkolban, 2017-02-06:
Following a great test suggestion from Sergey, I ran a "curl" test against the ESP32 server and the identical symptoms were produced.  As such, this puzzle is not limited to WebSockets but appears to be related to any HTTP connection request incoming.

nkolban, 2017-02-06:
Aha ... a test and a solution ...
In my event handler, I had the following coded:
```
char temp[100];
mg_conn_addr_to_str(nc, temp, sizeof(temp),
		MG_SOCK_STRINGIFY_IP | MG_SOCK_STRINGIFY_PORT | MG_SOCK_STRINGIFY_REMOTE);
```
The purpose of this fragment was to obtain the IP address and port of the partner for diagnostic logging.  When I removed these statements from my event handler, ALL now works as expected.  What this seems to tell me is that attempting to call the `mg_conn_addr_to_str` API within the context on an event handler is only allowable for certain events and states and should not be called arbitrarily for just ANY incoming event.

closed 2017-02-06 by nkolban

snahmad, 2018-09-11:
mg_mgr_poll(&mgr, ???);

How much poll timeout i need to specify good enough to handle few TCP/UDP connection and one HTTP server connection.

What this timeout means,

cpq, 2018-09-12:
this timeout is the time mg_mgr_poll() returns if there is no IO.
Practically, 1 second is good.

snahmad, 2018-09-12:
Thanks for reply I appreciate it.

I hope that mg_mgr_poll does not sleep if no work. 
I have list of messages of send via TCP connection.
I was doing mg_send within MG_EV_POLL event. I think it was cause delay on my Http Web server pages.
Now I do mg_send of list of message after  mg_mgr_poll  with 1ms within same thread. now my Esp32 is copying with high load.