cesanta/mongoose #783: Fails to build on Windows MinGW ("fileno redefined")

todbot, 2017-02-08:
Hi,
I've been struggling trying to get Mongoose to compile on Windows 10 w/ VS2015.

To demonstrate what I'm seeing. I created a very simple repo here:
  https://github.com/todbot/mongoose-tests
that is simply a holder for an Appveyor.yml file and a submodule.  

The build results on Appveyor are exactly what I'm seeing locally. I'm using Appveyor's standard install of MinGW and you can see how simple the appveyor.yml config file is.  You can see compilation errors I'm seeing here:
  https://ci.appveyor.com/project/todbot/mongoose-tests/build/1.0.5

Am I doing something wrong?

cpq, 2017-02-08:
Alex, please fix and use the similar setup to set mongoose on appveyor/mingw.

benrubson, 2017-02-28:
I'm facing exactly same issue and had to add the following to `mongoose.h` (line 212) :
```
#if defined(__MINGW32__)
#undef fileno
#undef SLIST_ENTRY
#include <stdint.h>
#include <ctype.h>
#endif
```
@mongoose devs, does this sound good ?
Thx 👍

benrubson, 2017-02-28:
btw, which Windows compiler do you usually use to build this project ?
Many thx 👍

cpq, 2017-03-01:
@alashkin friendly ping

benrubson, 2017-03-06:
No issue with the much more uptodate [MSYS2](http://www.msys2.org) 👍 
Perhaps we should simply discourage MinGW32 / MSYS (mingw.org), and advise MSYS2 ?

todbot, 2017-03-06:
Maybe I'm missing how to do this in MSYS2, but I cannot find a way to create an executable that doesn't depend on an external "msys-2.0.dll" file.  I'm trying to build a single executable with no non-Windows shared lib dependencies.

The original MSYS/MinGW can do this no problem, but I cannot figure out how to do this with MSYS2.

benrubson, 2017-03-06:
Install [MSYS2](http://www.msys2.org), i686 or x86_64, up to you, depending on your building host.

Using the "_MSYS2 MSYS_" start menu link :
- `pacman -Syu` #update packages, eventually twice if asked
- `pacman -S base-devel mingw-w64-<i686|x86_64>-toolchain tar` #install dev tools

Then launch the "_MSYS2 MinGW <32|64>-bits_" start menu link, whether you build a 32 or 64 bits app.
**And build your app** 👍

todbot, 2017-03-06:
Thanks @benrubson, this is extremely useful.  I had just figured out that perhaps doing "pacman -S gcc" is not the obviously right thing to do for MSYS2.

Have you experience with using MSYS2 on Appveyor?

benrubson, 2017-03-07:
U're welcome ! No, no experience with MSYS2 on Appveyor.

alashkin, 2017-03-30:
Added a couple of `#ifdef` in 57a4ec31d10d8b30a465354567b010ed726189f6
Should be OK now

closed 2017-03-30 by alashkin