cesanta/mongoose #784: lwIP + FreeRTOS support

PawelMag, 2017-02-13:
Hi!

I'm trying to run Mongoose on Kinetis K64 MCU with lwIP and FreeRTOS. I poll mg_mgr_poll() function in sinle FreeRTOS thread. This application is ok until i try to get index.html page on my pc. The lwIP task hangs at LWIP_ASSERT("tcp_receive: valid queue length"). I think that Mongoose is using lwIP only in NON-OS mode.
How to enable FreeRTOS support on Mongoose stack?

rojer, 2017-02-13:
in bare-metal mode mongoose uses low level LWIP api which does not require sockets task.
in os mode, if lwip is compiled with socket support, it is preferable to use the sockets interface.

change [this line](https://github.com/cesanta/mongoose/blob/fc113d6/mongoose.h#L1466) to `MG_NET_IF_SOCKET` and see if it works better.

cpq, 2017-03-15:
@PawelMag marking this as solved. feel free to reopen if it's still does not work for you.

closed 2017-03-15 by cpq