cesanta/mongoose #789: Issue when linking with OpenSSL

benrubson, 2017-03-02:
Hi,

Here is the issue I have when static-linking with OpenSSL (libcrypto) :

```
libcrypto.a(md5_dgst.o):md5_dgst.c:(.text+0x0): multiple definition of `MD5_Init'
mongoose.o:mongoose.c:(.text+0xa50): first defined here
libcrypto.a(md5_dgst.o):md5_dgst.c:(.text+0x7c0): multiple definition of `MD5_Update'
mongoose.o:mongoose.c:(.text+0x139e): first defined here
libcrypto.a(md5_dgst.o):md5_dgst.c:(.text+0xa40): multiple definition of `MD5_Final'
mongoose.o:mongoose.c:(.text+0x150b): first defined here
```

Thank you !

Ben

benrubson, 2017-03-02:
Solved by PR #790 👍

cpq, 2017-03-15:
Just defined `-DDISABLE_MD5` in your compilation flags.

closed 2017-03-15 by cpq

benrubson, 2017-03-15:
OK thx 👍

benrubson, 2017-03-15:
@cpq according to the doc, the flag is `MG_DISABLE_MD5` :
https://docs.cesanta.com/mongoose/master/#/overview/build-options/disabling-flags.md/
For according to source code, seems to be `DISABLE_MD5`.
Same for SHA1.
Is doc wrong ?
Thx 👍

benrubson, 2017-03-15:
Or perhaps the code should have to be updated adding the `MG_` prefix to the 3 flags ?
```
DISABLE_BASE64
DISABLE_MD5
DISABLE_SHA1
```

benrubson, 2017-03-16:
See #796 for this.

rhulha, 2017-05-17:
for me the solution to compile together with openssl was to use: CS_ENABLE_NATIVE_MD5