cesanta/mongoose #795: Handshake call using mg_send_websocket_handshake2 from a web socket client is not sent to the web server

tfgsupport, 2017-03-10:
Hi,

I am using Mongoose to send a handshake from a web socket client to a web server using the mg_send_websocket_handshake2 function. Below is my simple web server and simple client test cases. The mg_send_websocket_handshake2 call made by the server is not received by the client event handler as I expected. Why is this?

Thanks,

Nilufar

**Simple Web Socket Client**

```
#include "mongoose.h"

static int s_done = 0;
static int s_is_connected = 0;

static void ev_handler( struct mg_connection *nc, int ev, void *ev_data ) 
{
	struct websocket_message *wm = ( struct websocket_message * ) ev_data;
	(void)nc;

	switch( ev ) 
	{
		case MG_EV_CONNECT: 
		{
			int status = *( (int *)ev_data );
			if( status != 0 ) {
				printf( "-- Connection error: %d\n", status );
			}
			break;
		}
		case MG_EV_HTTP_REPLY:
		{
			struct http_message *hm = static_cast< struct http_message * >( ev_data );
			std::cout << "Received data (printf) from server : " << hm->message.p << std::endl;
			// CLIENT HANDSHAKE REQUEST
			std::string header( "Authorization: Auth Token\r\n" );
			mg_send_websocket_handshake2( nc, "/", "127.0.0.1:8001", NULL, header.c_str() );
			// CLIENT HANDSHAKE REQUEST
			break;
		}
		case MG_EV_WEBSOCKET_HANDSHAKE_DONE: {
			printf( "-- Connected\n" );
			s_is_connected = 1;
			break;
		}
		case MG_EV_WEBSOCKET_HANDSHAKE_REQUEST: {
			printf( "Web socket handshake request." );
			break;
		}
	}
}

BOOST_AUTO_TEST_CASE( WebSocketClientTest )
{
	struct mg_mgr mgr;
	struct mg_connection * connection;
	const char * url = "ws://127.0.0.1:8001";

	mg_mgr_init( &mgr, NULL );

	connection = mg_connect_ws( &mgr, ev_handler, url, "ws_chat", NULL );

	if( connection == NULL ) 
	{
		fprintf( stderr, "Invalid address\n" );
		return;
	}

	while( !s_done ) {
		mg_mgr_poll( &mgr, 100 );
	}
	mg_mgr_free( &mgr );
}
```

**Simple Web Server**

```
#include "mongoose.h"

static const char *s_http_port = "8001";
static struct mg_serve_http_opts s_http_server_opts;

static void ev_handler( struct mg_connection *nc, int ev, void * event_data ) {
	
	struct http_message *hm = static_cast< struct http_message * >( event_data );

	if( ev == MG_EV_HTTP_REQUEST ) {
		mg_serve_http( nc, ( struct http_message * ) event_data, s_http_server_opts );
		return;
	}
	else if( ev == MG_EV_WEBSOCKET_HANDSHAKE_REQUEST )
	{
		mg_printf( nc,
			"HTTP/1.1 401 Unauthorized\r\n"
			"WWW-Authenticate: NTLM\r\n"
			"WWW-Authenticate: Negotiate\r\n"
			"Connection: close\r\n"
			"Content-Type: text/html\r\n"
			"Content-Length: 0\r\n\r\n",
			"localhost", ( unsigned long ) time( NULL ) );
	}
}

BOOST_AUTO_TEST_CASE( WebSocketServerTest )
{
	struct mg_mgr mgr;
	struct mg_connection *nc;

	mg_mgr_init( &mgr, NULL );
	printf( "Starting web server on port %s\n", s_http_port );
	nc = mg_bind( &mgr, s_http_port, ev_handler );
	if( nc == NULL ) {
		printf( "Failed to create listener\n" );
	}

	// Set up HTTP server parameters
	mg_set_protocol_http_websocket( nc );
	s_http_server_opts.document_root = ".";  // Serve current directory
	s_http_server_opts.enable_directory_listing = "yes";

	for( ;; ) {
		mg_mgr_poll( &mgr, 1000 );
	}
	mg_mgr_free( &mgr );
}
```

cpq, 2017-03-10:
`mg_send_websocket_handshake2()` must be called by a client.

With Websocket, it is a client who initiates the handshake and upgrades the protocol, not the server.

tfgsupport, 2017-03-10:
I typed my question incorrectly. Please see the corrected question below.

The mg_send_websocket_handshake2 call made by the **client** is not received by the **server** event handler as I expected. Why is this?

tfgsupport, 2017-03-13:
We've noticed that the protocol handler switches from mg_http_handler to mg_websocket_handler after the initial MG_EV_WEBSOCKET_HANDSHAKE_REQUEST. The simple web server can then no longer parse the message from the mg_send_websocket_handshake2 call as it treats it as a web socket message and not a http message. We can set the handler back to http using mg_set_protocol_http_websocket() in the MG_EV_WEBSOCKET_HANDSHAKE_REQUEST case. Is this the correct way to do this?

alashkin, 2017-03-13:
Hi, @tfgsupport 

I just tried your code. 
I added a couple of "printfs" to server, like this:
```
	if( ev == MG_EV_HTTP_REQUEST ) {
        printf("MG_EV_HTTP_REQUEST\n");
....
	}
	else if( ev == MG_EV_WEBSOCKET_HANDSHAKE_REQUEST )
	{
        printf("MG_EV_WEBSOCKET_HANDSHAKE_REQUEST\n");
        ......		
```

My output:
```
[ ~/go/src/cesanta.com/mongoose @rpt_1 ] ./server 
Starting web server on port 8001
MG_EV_WEBSOCKET_HANDSHAKE_REQUEST
```
```
[ ~/go/src/cesanta.com/mongoose @rpt_1 ] ./client 
Received data (printf) from server : 0x6a22d0
```
You wrote:
>call made by the client is not received by the server 

But looks like it does receive. Or I misunderstood question?

Also, I notices this in your code:
``` 
		mg_printf( nc,
			"HTTP/1.1 401 Unauthorized\r\n"
			"WWW-Authenticate: NTLM\r\n"
			"WWW-Authenticate: Negotiate\r\n"
			"Connection: close\r\n"
			"Content-Type: text/html\r\n"
			"Content-Length: 0\r\n\r\n",
			"localhost", ( unsigned long ) time( NULL ) );
```
This code doesn't look correct.
mg_printf works like regular printf, i.e. second parameter should be _format_line with %s, %d etc.
In you case, these parameters
`			"localhost", ( unsigned long ) time( NULL ) );` are just ignored.

Is it by design/typo/something else?

tfgsupport, 2017-03-13:
Hi Alashkin,

The client connects initally using **`mg_connect_ws( &mgr, ev_handler, url, "ws_chat", NULL )`** and this is received by the web server.

Web server replies with HTTP 401 which the client receives. The client then sends a handshake request using **`mg_send_websocket_handshake2( nc, "/", "127.0.0.1:8001", NULL, header.c_str() );`** in the **`MG_EV_HTTP_REPLY`** case but this is **not** received by the web server.

When debugging the server, it can be seen that on **`MG_EV_WEBSOCKET_HANDSHAKE_REQUEST`** event, **`proto_handler`** for that connection is set to **`mg_websocket_handler`** and this in turn causes any subsequent messages on **`MG_EV_RECV`** event to be parsed under this handler. As you can see under the client code, we are attempting to send a 2nd HTTP message on **`MG_EV_HTTP_REPLY`** via **`mg_send_websocket_handshake2`** call. When this 2nd message arrives on the server, it is unable to parse it because **`mg_websocket_handler`** does not understand the HTTP message.

Our question is, should this handler be reset to **`mg_set_protocol_http_websocket`** on the server side on **`MG_EV_WEBSOCKET_HANDSHAKE_REQUEST`** so that it is able to parse the 2nd message? Is that the right thing to do?

What we are trying to do is to implement authentication on websocket handshake request. The idea is there will be several passes before the connection is authenticated hence this code structure.

Thank you. I will fix my **`mg_printf`** statement.

Nilufar

alashkin, 2017-03-13:
Seems Mongoose does as it has to - if protocol is WS - the protocol is WS

Maybe, you should connect using mg_connect_http and only after all HTTP tasks are done switch to WebSocket?

cpq, 2017-03-14:
@tfgsupport so it's only after a long conversation you point out the real intention - implementing authentication mechanism. It would save us some round trips if you did that from the beginning.

As Alex has mentioned, as soon as you're triggering protocol upgrade, the connection is under the websocket event handler, and this is by design. Mongoose behaves as expected.

Any reasons you don't use TLS for authentication?

alashkin, 2017-03-21:
@cpq 
Reassigning to you. Feel free to return it back to me

tfgsupport, 2017-03-21:
@alashkin @cpq Thanks for the explanation.

We would like to implement authentication from a C++ client similar to the browser. Sorry that was not clear.

For authenticating with the **browser** we **close the connection** after the **401 response** to allow the mongoose web socket handling to re-initialise and allow another **MG_EV_WEBSOCKET_HANDSHAKE_REQUEST** as discussed in #750.

For authenticating with the **C++ client** we believe we have to reset the **proto_handler** to **mg_http_handler** after the **401 response** to allow another **MG_EV_WEBSOCKET_HANDSHAKE_REQUEST**.

We are investigating TLS for authentication.

cpq, 2017-03-23:
Thank you @tfgsupport !
Can we close this then?
We can help with the TLS option, would you be interested?

tfgsupport, 2017-03-24:
Yes this can be closed.

We'll consider this in the future and come back if we need help.

closed 2017-04-02 by cpq