cesanta/mongoose #797: cleanSession flag for MQTT connection?

forthlightning, 2017-03-30:
Hello! Does the mqtt section include cleanSession flag? 

I would like to set cleanSession to false if this is possible.

rojer, 2017-03-30:
`mos config-set mqtt.clean_session=false`

rojer, 2017-03-30:
ah, sorry, it's fro mongoose, not mongoose-os... yes, there is a flag in `mg_send_mqtt_handshake_opts.flags`. however, it defaults to false, so if you do not set `opts.flags |= MG_MQTT_CLEAN_SESSION`, it won't be set to broker.

rojer, 2017-03-30:
see example of mongoose mqtt client connection setup (in mOS) [here](https://github.com/cesanta/mongoose-os/blob/a0ab23402ef7b878d55d3fc21a613721293847dd/fw/src/mgos_mqtt.c#L217).

let me know if this answers your question.

forthlightning, 2017-03-30:
dang that was quick! Thanks! 

I just had a look around the docs and was going to do something like 

```
uint8_t my_flags = 0x00;
my_flags |= MG_MQTT_CLEAN_SESSION;
```

and then feed my_flags into opts.

But it sounds like there are some nice structs to set it all up, i'll try those out!

closed 2017-03-30 by forthlightning