cesanta/mongoose #799: Issues when using mqtt-client with mbedtls

anyn99, 2017-04-08:
Hi there,

i can't get the mqtt-client example running when using ssl with mbedtls.

Here my altered example with mbedtls:

`static const char *s_address = "192.168.0.7:8883";
static const char *s_user_name = "username";
static const char *s_password = "password";
static const char *s_topic = "/stuff";

static const char *cipher_suites =
		"TLS-RSA-WITH-AES-128-GCM-SHA256:"
        "TLS-RSA-WITH-AES-128-CBC-SHA256:"
		"TLS-RSA-WITH-AES-128-CBC-SHA";

static struct mg_mqtt_topic_expression s_topic_expr = {NULL, 0};

static void MqttEventHandler(struct mg_connection *nc, int ev, void *p) {
  struct mg_mqtt_message *msg = (struct mg_mqtt_message *) p;
  (void) nc;

#if 1
  if (ev != MG_EV_POLL)
    DEBUG_INFO("USER HANDLER GOT %d\n", ev);
#endif

  switch (ev) {
    case MG_EV_CONNECT: {
      int connect_status = * (int *) p;
      if (connect_status != 0)
      {
    	  DEBUG_ERROR("Error: %d (%s)\r\n", connect_status, strerror(connect_status));
      }
      else
      {
    	  DEBUG_INFO("Got connected\r\n");
      }

      struct mg_send_mqtt_handshake_opts opts;
      memset(&opts, 0, sizeof(opts));
      opts.user_name = s_user_name;
      opts.password = s_password;

      mg_set_protocol_mqtt(nc);
      mg_send_mqtt_handshake_opt(nc, "Test-Esp", opts);
      break;
    }
    case MG_EV_MQTT_CONNACK:
      if (msg->connack_ret_code != MG_EV_MQTT_CONNACK_ACCEPTED) {
        DEBUG_INFO("Got mqtt connection error: %d\n", msg->connack_ret_code);
        break;
      }
      s_topic_expr.topic = s_topic;
      s_topic_expr.qos = 0;
      DEBUG_INFO("Subscribing to '%s'\r\n", s_topic);
      mg_mqtt_subscribe(nc, &s_topic_expr, 1, 42);
      break;
    case MG_EV_MQTT_PUBACK:
      DEBUG_INFO("Message publishing acknowledged (msg_id: %d)\n", msg->message_id);
      break;
    case MG_EV_MQTT_SUBACK:
      DEBUG_INFO("Subscription acknowledged, forwarding to '/test'\n");
      break;
    case MG_EV_MQTT_PUBLISH: {
#if 0
        char hex[1024] = {0};
        mg_hexdump(nc->recv_mbuf.buf, msg->payload.len, hex, sizeof(hex));
        DEBUG_INFO("Got incoming message %.*s:\n%s", (int)msg->topic.len, msg->topic.p, hex);
#else
      DEBUG_INFO("Got incoming message %.*s: %.*s\n", (int) msg->topic.len,
             msg->topic.p, (int) msg->payload.len, msg->payload.p);
#endif

      DEBUG_INFO("Forwarding to /test\n");
      mg_mqtt_publish(nc, "/test", 65, MG_MQTT_QOS(0), msg->payload.p,
                      msg->payload.len);
      break;
    }
    case MG_EV_CLOSE:
      DEBUG_INFO("Connection closed\n");
      break;
  }
}

static void MqttTask(void *arg)
{
	struct mg_mgr mgr;

	DEBUG_INFO("MqttTask starting\r\n");

	mg_mgr_init(&mgr, NULL);

	struct mg_connect_opts opts;
	memset(&opts, 0, sizeof(opts));

	opts.ssl_ca_cert = "ca.crt";
	opts.ssl_cipher_suites = cipher_suites;

	DEBUG_INFO("Mongoose Mqtt client starting\r\n");

	if (mg_connect_opt(&mgr, s_address, MqttEventHandler, opts) == NULL)
	{
		DEBUG_ERROR("mg_connect(%s) failed\r\n", s_address);
		vTaskDelete(NULL);
	}

	while (1)
	{
		mg_mgr_poll(&mgr, 1000);
	}

}

void MqttInit(void)
{
	DEBUG_INFO("MqttInit\r\n");

	xTaskCreate(MqttTask, "Mongoose_Mqtt",
						  configMINIMAL_STACK_SIZE * 6,
			              NULL, 2, NULL);
}`

This example successfully executes until event MG_EV_MQTT_CONNACK,
the last line i get is "Subscribing to /stuff", but the server never gets any subscription request.
The last line in the server log says it was successfully sending CONNACK to client, then, after a timeout, it disconnects the client.

Am i missing anything? The ssl connection seems to be fully working, otherwise i wouldn't get the MG_EV_CONNECT, MG_EV_SEND, MG_EV_RECV and MG_EV_MQTT_CONNACK events.
But the very same code works without ssl encryption.

I am compiling for the esp8266, with esp-open-rtos (https://github.com/SuperHouse/esp-open-rtos)
with following flags:

'#define CS_PLATFORM CS_P_ESP8266
#define RTOS_SDK

#define MG_ENABLE_MQTT			1
#define MG_ENABLE_HTTP_SSI 		1
#define MG_ENABLE_FILESYSTEM	1
#define MG_ENABLE_SSL			1

#define MG_SSL_IF 	MG_SSL_IF_MBEDTLS

#define MG_ENABLE_DEBUG			1'

rojer, 2017-04-08:
i see you enabled debug. can you post the debug log of the connection?

anyn99, 2017-04-09:
Hi!
Sure! Thanks for the quick answer :-)
There are also my own debug messages in the log, this is the output of above code:
```
WifiTask             (wifi_manager.c):   61 [INFO] : WiFi: Connected
MqttTask             (mqtt.c):  133 [INFO] : Mongoose Mqtt client starting
mg_connect_opt       0x3fff4910 192.168.0.17:8883 -,-,ca.crt
mg_ssl_if_conn_init  0x3fff4910 ,,ca.crt
mg_set_cipher_list   TLS-RSA-WITH-AES-128-GCM-SHA256 -> 009c
mg_set_cipher_list   TLS-RSA-WITH-AES-128-CBC-SHA256 -> 003c
mg_set_cipher_list   TLS-RSA-WITH-AES-128-CBC-SHA -> 002f
mg_do_connect        0x3fff4910 tcp://192.168.0.7:8883
mg_socket_if_connect 0x3fff4910 sock 0 rc 0 errno 0 err 0
mg_add_conn          0x3fff8bdc 0x3fff4910
MqttTask             (mqtt.c):  141 [INFO] : Mg_connect ok
mg_mgr_handle_conn   0x3fff4910 fd=0 fd_flags=2 nc_flags=24 rmbl=0 smbl=0
ssl_socket_send      0x3fff4910 143 -> 143
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 53 <- 53
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 2269 <- 1397
ssl_socket_recv      0x3fff4910 872 <- 872
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 4 <- 4
ssl_socket_send      0x3fff4910 267 -> 267
ssl_socket_send      0x3fff4910 6 -> 6
ssl_socket_send      0x3fff4910 45 -> 45
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 186 <- 186
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 1 <- 1
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 40 <- 40
mg_ssl_begin         0x3fff4910 0 res 0
mg_if_connect_cb     0x3fff4910 connect, err=0
mg_call              0x3fff4910 user ev=2 ev_data=0x3fff8650 flags=48 rmbl=0 smbl=0
MqttEventHandler     (mqtt.c):   43 [INFO] : USER HANDLER GOT 2
MqttEventHandler     (mqtt.c):   55 [INFO] : Got connected
mg_call              0x3fff4910 after user flags=48 rmbl=0 smbl=57
ssl_socket_send      0x3fff4910 86 -> 86
mg_ssl_if_write      0x3fff4910 57 -> 57
mg_write_to_socket   0x3fff4910 57 bytes -> 0 (SSL)
mg_call              0x3fff4910 proto ev=4 ev_data=0x3fff8680 flags=48 rmbl=0 smbl=0
MqttEventHandler     (mqtt.c):   43 [INFO] : USER HANDLER GOT 4
mg_call              0x3fff4910 after proto flags=48 rmbl=0 smbl=0
mg_mgr_handle_conn   0x3fff4910 after fd=0 nc_flags=48 rmbl=0 smbl=0
mg_mgr_handle_conn   0x3fff4910 fd=0 fd_flags=1 nc_flags=48 rmbl=0 smbl=0
ssl_socket_recv      0x3fff4910 5 <- 5
ssl_socket_recv      0x3fff4910 28 <- 28
mg_ssl_if_read       0x3fff4910 1024 -> 4
mg_handle_tcp_read   0x3fff4910 4 bytes <- 0 (SSL)
mg_recv_common       0x3fff4910 4 0
mg_call              0x3fff4910 proto ev=3 ev_data=0x3fff8670 flags=48 rmbl=4 smbl=0
MqttEventHandler     (mqtt.c):   43 [INFO] : USER HANDLER GOT 3
MqttEventHandler     (mqtt.c):   43 [INFO] : USER HANDLER GOT 202
MqttEventHandler     (mqtt.c):   81 [INFO] : Subscribed to '/bla'
mg_call              0x3fff4910 after proto flags=48 rmbl=0 smbl=13
```

for completeness, the log of the mosquitto-mqtt broker i am connecting with:
```
1491772943: mosquitto version 1.4.11 (build date 20/02/2017 23:24:29.40) starting
1491772943: Config loaded from mosquitto.conf.
1491772943: Opening ipv6 listen socket on port 8883.
1491772943: Opening ipv4 listen socket on port 8883.
1491772959: New connection from 192.168.0.10 on port 8883.
1491772959: New client connected from 192.168.0.10 as Test-Esp (c1, k120, u'esp').
1491772959: Sending CONNACK to Test-Esp (0, 0)
```

anyn99, 2017-04-10:
Another interesting thing i recognized:

After the above output, lwip also stalls in sys_arch_mbox_fetch(), which is the platfrom specific implementation of sys_mbox_fetch(). The thread blocks here until the broker sends that he is quitting the connection due to keep alive timeout.

So it might actually not be an mongoose issue... but the question is, why does lwip end with waiting for tcp input, instead of sending something out? the send mbuf length (smbl=13) stays like that all the time, so is actually never sent out.
As stated before, everything works fine with ssl diabled.

anyn99, 2017-04-12:
ok, got the bug.
The "while ((n = mg_ssl_if_read(conn, buf, MG_TCP_RECV_BUFFER_SIZE)) > 0)" inside mg_handle_tcp_read()
is deadly when used with mbedtls. The called function mbedtls_ssl_read() is waiting forever to receive a ssl header of 5 bytes. I don't know yet why it is not returning 0 when reaching end of record.

anyn99, 2017-04-17:
Hi again.
After a lot of digging into mongoose, mbedtls and lwip, i can't say for sure that this behaviour is caused by faulty code inside mongoose.
Here are my findings:
If mg_handle_tcp_read() is called, it never returns when using mbedtls.
The reason for this is: mbedtls_ssl_read() ( called by mg_ssl_if_read() ) never returns 0.
Therefore this loop:  while ((n = mg_ssl_if_read(conn, buf, MG_TCP_RECV_BUFFER_SIZE)) > 0) never exits.

I followed the function calls all through the mbedtls library into lwip. In the end the lwip implementation i am using (the one from esp-open-rtos) ends in calling sys_arch_mbox_fetch(), which just stalls waiting for incoming data. 

Here is what i did to get the example running again. I replaced this code inside mg_handle_tcp_read():
```
#if MG_ENABLE_SSL
  if (conn->flags & MG_F_SSL) {
    if (conn->flags & MG_F_SSL_HANDSHAKE_DONE) {
      /* SSL library may have more bytes ready to read than we ask to read.
       * Therefore, read in a loop until we read everything. Without the loop,
       * we skip to the next select() cycle which can just timeout. */
      while ((n = mg_ssl_if_read(conn, buf, MG_TCP_RECV_BUFFER_SIZE)) > 0) {
        DBG(("%p %d bytes <- %d (SSL)", conn, n, conn->sock));
        mg_if_recv_tcp_cb(conn, buf, n, 1 /* own */);
        buf = NULL;
        if (conn->flags & MG_F_CLOSE_IMMEDIATELY) break;
        /* buf has been freed, we need a new one. */
        buf = (char *) MG_MALLOC(MG_TCP_RECV_BUFFER_SIZE);
        if (buf == NULL) break;
      }
      MG_FREE(buf);
      if (n < 0 && n != MG_SSL_WANT_READ) conn->flags |= MG_F_CLOSE_IMMEDIATELY;
    } else {
      MG_FREE(buf);
      mg_ssl_begin(conn);
      return;
    }
  } else
#endif
```
with this:
```
#if MG_ENABLE_SSL
  if (conn->flags & MG_F_SSL) {
    if (conn->flags & MG_F_SSL_HANDSHAKE_DONE) {
      n = mg_ssl_if_read(conn, buf, MG_TCP_RECV_BUFFER_SIZE);
      DBG(("%p %d bytes <- %d (SSL)", conn, n, conn->sock));
      if (n > 0)
      {
    	  mg_if_recv_tcp_cb(conn, buf, n, 1 /* own */);
    	  printf("after mg_if_recv_tcp_cb\r\n");
      }
      else
      {
    	  MG_FREE(buf);
      }
      if (n == 0) {
            /* Orderly shutdown of the socket, try flushing output. */
            conn->flags |= MG_F_SEND_AND_CLOSE;
      }
      if (n < 0) conn->flags |= MG_F_CLOSE_IMMEDIATELY;
    } else {
      MG_FREE(buf);
      mg_ssl_begin(conn);
      return;
    }
  } else
#endif
```

This is clearly a hack and not a solution, if the incoming data is bigger than MG_TCP_RECV_BUFFER_SIZE it will fail.

Can anyone help? Are there no issues like this if the Espressif RTOS-SDK for ESP8266 is being used?
And by the way, where do all those obviously unused MG_ESP8266 Precompiler clauses come from?

anyn99, 2017-04-22:
In addition to the last post i added
```
 #define MG_SSL_IF_MBEDTLS_MAX_FRAG_LEN 1024
```
To match tcp max buffer size.

As long as the server support max fragmentation to be set, it should work like this.

rojer, 2017-04-22:
sorry for delayed response. let me address issues i see here.
first, blocking i/o - this should never happen. read should return EWOULDBLOCK. somehow socket was not put into non-blocking mode?
i seem to recall that we disable non-blocking mode on esp8266 with sockets... which may be wrong.
the reason we do not hit this in mongoose-os is because on esp8266, non-os and rtos, we use low-level lwip handler, not using sockets.
i'll try esp8266 + sockets on rtos and report. it should be fixed. but in teh mean time i suggest you actually try lwip low_level net_if, as it seems to work ok and will use slightly less ram.

sky9971, 2019-11-14:
hi,
I'm also trying the mqtt client by mongoose using ssl in rtos.
but in my case it is giving a compile time error.
If I set SSL to OPENSSL it throws:

`C:/msys32/home/jack/esp/ESP8266_RTOS_SDK/examples/mg-mqtt/components/mongoose/mongoose.c:4672:26: fatal error: openssl/tls1.h: No such file or directory`

for Mbedtls it gives :

```
C:/msys32/home/jack/esp/ESP8266_RTOS_SDK/examples/mg-mqtt/components/mongoose/mongoose.c: In function 'mg_use_ca_cert':
C:/msys32/home/jack/esp/ESP8266_RTOS_SDK/examples/mg-mqtt/components/mongoose/mongoose.c:5395:7: error: implicit declaration of function 'mbedtls_x509_crt_parse_file' [-Werror=implicit-function-declaration]
   if (mbedtls_x509_crt_parse_file(ctx->ca_cert, ca_cert) != 0) {
       ^
C:/msys32/home/jack/esp/ESP8266_RTOS_SDK/examples/mg-mqtt/components/mongoose/mongoose.c: In function 'mg_use_cert':
C:/msys32/home/jack/esp/ESP8266_RTOS_SDK/examples/mg-mqtt/components/mongoose/mongoose.c:5419:7: error: implicit declaration of function 'mbedtls_pk_parse_keyfile' [-Werror=implicit-function-declaration]
   if (mbedtls_pk_parse_keyfile(ctx->key, key, NULL) != 0) {
```

Any help will be appreciated. Thanks in advance.

cpq, 2020-12-13:
Please use the latest code, there is MQTT client example at https://github.com/cesanta/mongoose/tree/master/examples/mqtt-client

closed 2020-12-13 by cpq