cesanta/mongoose #804: Issue during building for ESP8266

r44083, 2017-04-28:
Hi, when I tried to build simple mongoose example for ESP8266 I faced with errors during linking.
This errors looks like they related to stdlib.
See my build output and makefile below:
<details>
<p>
    <summary><small>My build output</summary>

```
- building ESP8266-rtos-hal...
make clean
make[1]: Entering directory `F:/electronics/sw/proj/sublimetext-projects/simple_hal/ESP8266-rtos-hal'
- cleaning out/obj, out/lst, out/bin...
(out/obj:&(rd /s /q "out/obj" 2> NUL))&
(out/lst:&(rd /s /q "out/lst" 2> NUL))&
(out/bin:&(rd /s /q "out/bin" 2> NUL))&
make[1]: Leaving directory `F:/electronics/sw/proj/sublimetext-projects/simple_hal/ESP8266-rtos-hal'
make dirs
make[1]: Entering directory `F:/electronics/sw/proj/sublimetext-projects/simple_hal/ESP8266-rtos-hal'
- making out/obj, out/lst, out/bin...
(out/obj:&(mkdir "out/obj" 2> NUL))&
(out/lst:&(mkdir "out/lst" 2> NUL))&
(out/bin:&(mkdir "out/bin" 2> NUL))&
make[1]: Leaving directory `F:/electronics/sw/proj/sublimetext-projects/simple_hal/ESP8266-rtos-hal'
make out/bin/ESP8266-rtos-hal.elf
make[1]: Entering directory `F:/electronics/sw/proj/sublimetext-projects/simple_hal/ESP8266-rtos-hal'
- compiling src/main.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isrc/hal/ESP8266_SDK/include/lwip -Isrc/hal/ESP8266_SDK/include/lwip/ipv4 -Isrc/hal/ESP8266_SDK/include/lwip/ipv6 -Isrc/hal/ESP8266_SDK/include/mbedtls -Isrc/hal/ESP8266_SDK/include/nopoll -Isrc/hal/ESP8266_SDK/include/openssl -Isrc/hal/ESP8266_SDK/include/spiffs -Isrc/hal/ESP8266_SDK/include/ssl -Os -g -std=gnu99 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -mno-serialize-volatile -c src/main.c -o out/obj/main.o
- compiling src/common/macros.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isrc/hal/ESP8266_SDK/include/lwip -Isrc/hal/ESP8266_SDK/include/lwip/ipv4 -Isrc/hal/ESP8266_SDK/include/lwip/ipv6 -Isrc/hal/ESP8266_SDK/include/mbedtls -Isrc/hal/ESP8266_SDK/include/nopoll -Isrc/hal/ESP8266_SDK/include/openssl -Isrc/hal/ESP8266_SDK/include/spiffs -Isrc/hal/ESP8266_SDK/include/ssl -Os -g -std=gnu99 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -mno-serialize-volatile -c src/common/macros.c -o out/obj/macros.o
- compiling src/drv/drv_di.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isrc/hal/ESP8266_SDK/include/lwip -Isrc/hal/ESP8266_SDK/include/lwip/ipv4 -Isrc/hal/ESP8266_SDK/include/lwip/ipv6 -Isrc/hal/ESP8266_SDK/include/mbedtls -Isrc/hal/ESP8266_SDK/include/nopoll -Isrc/hal/ESP8266_SDK/include/openssl -Isrc/hal/ESP8266_SDK/include/spiffs -Isrc/hal/ESP8266_SDK/include/ssl -Os -g -std=gnu99 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -mno-serialize-volatile -c src/drv/drv_di.c -o out/obj/drv_di.o
- compiling src/extra-lib/xprintf/xprintf.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isrc/hal/ESP8266_SDK/include/lwip -Isrc/hal/ESP8266_SDK/include/lwip/ipv4 -Isrc/hal/ESP8266_SDK/include/lwip/ipv6 -Isrc/hal/ESP8266_SDK/include/mbedtls -Isrc/hal/ESP8266_SDK/include/nopoll -Isrc/hal/ESP8266_SDK/include/openssl -Isrc/hal/ESP8266_SDK/include/spiffs -Isrc/hal/ESP8266_SDK/include/ssl -Os -g -std=gnu99 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -mno-serialize-volatile -c src/extra-lib/xprintf/xprintf.c -o out/obj/xprintf.o
- compiling src/extra-lib/mongoose/esp_libc.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isrc/hal/ESP8266_SDK/include/lwip -Isrc/hal/ESP8266_SDK/include/lwip/ipv4 -Isrc/hal/ESP8266_SDK/include/lwip/ipv6 -Isrc/hal/ESP8266_SDK/include/mbedtls -Isrc/hal/ESP8266_SDK/include/nopoll -Isrc/hal/ESP8266_SDK/include/openssl -Isrc/hal/ESP8266_SDK/include/spiffs -Isrc/hal/ESP8266_SDK/include/ssl -Os -g -std=gnu99 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -mno-serialize-volatile -c src/extra-lib/mongoose/esp_libc.c -o out/obj/esp_libc.o
- compiling src/extra-lib/mongoose/mongoose.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isrc/hal/ESP8266_SDK/include/lwip -Isrc/hal/ESP8266_SDK/include/lwip/ipv4 -Isrc/hal/ESP8266_SDK/include/lwip/ipv6 -Isrc/hal/ESP8266_SDK/include/mbedtls -Isrc/hal/ESP8266_SDK/include/nopoll -Isrc/hal/ESP8266_SDK/include/openssl -Isrc/hal/ESP8266_SDK/include/spiffs -Isrc/hal/ESP8266_SDK/include/ssl -Os -g -std=gnu99 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -mno-serialize-volatile -c src/extra-lib/mongoose/mongoose.c -o out/obj/mongoose.o
- compiling src/hal/hal_gpio.c...
xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DRTOS_SDK -DMG_ESP8266 -DCS_PLATFORM=CS_P_ESP8266 -Isrc -Isrc/common -Isrc/drv -Isrc/extra-lib -Isrc/extra-lib/xprintf -Isrc/extra-lib/mongoose -Isrc/hal -Isrc/hal/ESP8266_SDK/extra_include -Isrc/hal/ESP8266_SDK/extra_include/xtensa -Isrc/hal/ESP8266_SDK/extra_include/xtensa/config -Isrc/hal/ESP8266_SDK/extra_include/xtensa/tie -Isrc/hal/ESP8266_SDK/include -Isrc/hal/ESP8266_SDK/include/espressif -Isrc/hal/ESP8266_SDK/include/espressif/esp8266 -Isrc/hal/ESP8266_SDK/include/freertos -Isrc/hal/ESP8266_SDK/include/json -Isr [... cut ...]
F:\electronics\sw\proj\sublimetext-projects\simple_hal\ESP8266-rtos-hal/src/extra-lib/mongoose/mongoose.c:300: undefined reference to `vfprintf'
[... 288 lines cut ...]
	$(LD) $(LDFLAGS) $^ -o $@

$(BIN): $(ELF)
	@echo - making bin from $<...
	@esptool elf2image $< -o $(BINDIR)/$(TARGET)-

$(LSS): $(ELF)
	@echo - disassembling $<...
	$(OBJDUMP) -dC $< >> $@

$(OBJDIR)/%.o: %.c
	@echo - compiling $<...
	$(CC) $(CFLAGS) -c $< -o $@

$(OBJDIR)/%.o: %.cpp
	@echo - compiling $<...
	$(CPP) $(CPPFLAGS) -c $< -o $@

$(OBJDIR)/%.o: %.s
	@echo - compiling $<...
	$(AS) $(AFLAGS) -c $< -o $@
```

</p>
</details>

anyn99, 2017-07-13:
i think the main problem is you are building with the original SDK from espressif.
Would you mind using esp-open-sdk or esp-open-rtos?
i think you need full libgcc and not just the buggy espressif stuff.

r44083, 2019-02-21:
esp-open-sdk and esp-open-rtos are almost dead, whereas original ESP8266_RTOS_SDK is tending to regular releases and fixes.

closed 2020-04-05 by r44083