cesanta/mongoose #807: Question: Can/Should I use multiple mg_mgr in one application?

sgiurgiu, 2017-05-13:
I have an http server application. I also want to do DNS queries from time to time (as a client). Can I use different "struct mg_mgr" for the different roles (http server and dns client)? Or should I just create one single mg_mgr per application, and just use that for the different connections/roles?

Thank you.

PuKathy, 2018-11-07:
Multi mg_mgr works, but is not necessary.

struct mg_mgr mgr;
mg_mgr_init(&mgr);
struct mg_connection *nc_http_server = mg_connect_http(&mgr, ...
struct mg_connection *nc_dns_server = mg_connect_dns(&mgr, ...

cpq, 2020-12-13:
Closing this.

As was correctly mentioned, a single manager can handle multiple connections, and different protocols.

closed 2020-12-13 by cpq