cesanta/mongoose #812: Support http and https in one server

eggangel, 2017-05-17:
I am using the newest mongoose web server in my project. The demo in examples folder only contains simplest_web_server and simplest_web_server_ssl. But how to support http and https in one files. Please give me the steps. Thank you!

eggangel, 2017-05-17:
I have read the issue --- https://github.com/cesanta/mongoose/pull/385 . And I have tried "static char *s_http_port = "8001,ssl://443";". But only https port 8001 can be connected.

eggangel, 2017-05-17:
The main codes shows below.

static char *s_http_port = "8001,ssl://443";
static struct mg_serve_http_opts s_http_server_opts;
static const char *s_ssl_cert = "key/cert.pem";
static const char *s_ssl_key = "key/key.pem";

struct mg_mgr mgr;
struct mg_connection *nc;
struct mg_bind_opts bind_opts;
const char *err;
mg_mgr_init(&mgr, NULL);
memset(&bind_opts, 0, sizeof(bind_opts));
bind_opts.ssl_cert = s_ssl_cert;
bind_opts.ssl_key = s_ssl_key;
bind_opts.error_string = &err;

nc = mg_bind_opt(&mgr, s_http_port, ev_handler, bind_opts);
if (nc == NULL) {
	printf("Failed to create listener: %s\n", err);
	return 1;
}

mg_set_protocol_http_websocket(nc);
char * buffer = new char[2048];
if (!_getcwd(buffer, 2048))
	return 2;
strcat(buffer, "\\www\\");
string path = buffer;
delete buffer;
path = replace_all(path, "\\", "/");

s_http_server_opts.document_root = path.c_str();  // Serve current directory
s_http_server_opts.enable_directory_listing = "no";    //Set if show dir
	 
for (;;) {
	mg_mgr_poll(&mgr, 1000);
}
mg_mgr_free(&mgr);

eggangel, 2017-05-22:
@cpq Sorry to bother you. Could you please give some help on this issue? Thank you!

eggangel, 2017-08-19:
Find the solution. Thanks.

closed 2017-08-19 by eggangel

malotian, 2018-05-28:
I am also looking for a similar solution

malotian, 2018-06-13:
you can create NON-SSL server via configuration (port 80), 
and then SSL (443) Server programmatically, and provide default handler (NON-SSL)

    struct mg_bind_opts opts;
    opts.ssl_cert = "cert.pem";
    opts.ssl_key = "keys.pem";
    memset(&opts, 0, sizeof(opts));

    mg_set_protocol_http_websocket(mg_bind_opt(mgos_get_mgr(), "443", mgos_get_sys_http_server()->handler, NULL, opts));

Am yet to test this, But you can test before me and update status here.

pinkra, 2018-06-13:
@malotian thank you for your answer.
I've already found a solution, I call the mg_bind_opt and mg_set_protocol_http_websocket for each port :

```
    mg_mgr_init(&mgr, NULL);                                                                                           
    memset(&bind_opts, 0, sizeof(bind_opts));                                                                          
    bind_opts.user_data = my_data;                                                                                     
    nc_http = mg_bind_opt(&mgr, http_port, ev_handler, bind_opts);                                                     
#if HAVE_LIBSSL                                                                                                        
    bind_opts.ssl_cert = my_data->config->ssl_cert;                                                                    
    bind_opts.ssl_key = my_data->config->ssl_key;                                                                      
#endif                                                                                                                 
    nc = mg_bind_opt(&mgr, my_data->wai_port, ev_handler, bind_opts);
    mg_set_protocol_http_websocket(nc);                                                                                
    mg_set_protocol_http_websocket(nc_http);                                                                           
    s_http_server_opts.enable_directory_listing = "no";                                                                
    s_http_server_opts.document_root = WWWDIR;
```

dev31337, 2023-10-02:
In 2023, with master, anyone knows how to do this ?

scaprile, 2023-10-02:
Please do not multicast your issues
#2400 #2401