cesanta/mongoose #814: http / https

rollen1960, 2017-05-24:
I use mongoose with port 80 (HTTP) and port 443 (HTTPS). That's ok ... works fine.

I want to use the same port (for example 523) for accept HTTP and HTTPS request. It's possible ?

Thank you.

Mario

eggangel, 2017-05-25:
How to use mongoose with port 80 (HTTP) and port 443 (HTTPS) ?

rollen1960, 2017-05-25:
I write a server that open port 80 for HTTP and a port 443 for HTTPS with mongoose. That's work.
But ... my server must use the same port (for example 523) for respond to request in HTTP and request in HTTPS. Is that possible ? How ?
Thank you. Mario

Andrei-Paul, 2017-05-25:
Not sure if mongoose can do that, but this is a 'weird' request anyway. 
Maybe you can use HAProxy to determine what type of traffic comes at port 523 and forward the data to your local 80 and 443 for mongoose as needed. The "client" will not know the difference and the "server" need not be concerned with how the data gets here.
Instructions here:
http://timjrobinson.com/haproxy-how-to-run-http-https-on-the-same-port/

DISCLAIMER: I did not test if this actually works.

rollen1960, 2017-05-25:
Ok. I think is impossible because mongoose with SSL initialization will always decrypt the input stream. 
Thank you.
P.S. Tell me if you make a enhancement

closed 2017-05-25 by rollen1960