cesanta/mongoose #817: using connect opts

rollen1960, 2017-05-31:
I am writing a client with http and / or ws protocols.

If I use
		c = mg_connect_http(Mongoose_mgr, Mongoose_ev_handler, url.c_str(), headers.c_str(), request.c_str());

work but if I use

	        struct mg_connect_opts opts;
	        memset(&opts, '\0', sizeof(struct mg_connect_opts));
		c = mg_connect_http_opt(Mongoose_mgr, Mongoose_ev_handler, opts, url.c_str(), headers.c_str(), request.c_str());

it crash.

Why ?

PpvAlx, 2017-06-08:
Try to set mg_connect_opts.error_string.

rollen1960, 2017-06-09:
I probe:

        struct mg_connect_opts opts;
        memset(&opts, '\0', sizeof(struct mg_connect_opts));
        const char* error;
        opts.error_string = &error;
	c = mg_connect_http_opt(Mongoose_mgr, Mongoose_ev_handler, opts, url.c_str(), headers.c_str(), request.c_str());

but I receive c == NULL 

Why ? How can I read the error message ?

dimonomid, 2018-02-02:
You can read the error message by printing it:
```
printf("%s", error);
```

rollen1960, 2018-02-05:
I can no read the message because it crash in the mg_connect_http_opt routine.

dimonomid, 2018-02-06:
Previously you said

> but I receive c == NULL

So I assumed that `mg_connect_http_opt` actually returns.

cpq, 2020-12-14:
Closing this.

Note that the API has changed since 7.0 release.

closed 2020-12-14 by cpq