cesanta/mongoose #818: Parameters in reverse proxy

mortegai, 2017-05-31:
In a reverse proxy (either with the _mg_http_reverse_proxy_ function method or with the _url_rewrites_ option method) the request parameters (such as parm1 and parm2 in /api?parm1=value1&parm2=value2) are not passed to the backend server.

What is the way to do it?

cpq, 2017-07-20:
I believe https://github.com/cesanta/mongoose/commit/fea648496cf134a1aedddd81ee35bbc05e39853a fixes it.

closed 2017-07-20 by cpq

mortegai, 2017-07-20:
I don't think so. Tests do not work.
I think something must be done in:

7316    / * Send request upstream * /
 7317  mg_printf(be, "%.*s %.*s HTTP/1.1\r\n", (int) hm->method.len, hm->method.p,
  7318                   (int) path.len, path.p);