cesanta/mongoose #821: Documentation on user_data in mg_connection

nkolban, 2017-06-04:
I'm afraid I'm not getting what I was hoping for from the documentation of `struct mg_connection` with respect to `user_data`.  See:

https://docs.cesanta.com/mongoose/master/#/c-api/net.h/struct_mg_connection.md/

While I believe I understand the value of `user_data` as a place to hang my own context data, it isn't clear to me the scope of this data.  Specifically, if I have two parallel incomming connections, does each mg_connection have its "own" `user_data` pointer reference?

Let me try and explain further.

When I create an mgConnection using mg_bind(), I specify a value for `user_data` to point to some data.  When an incoming request callback event is invoked, I find that on THAT connection, the mg_connection's `user_data` is set to the value I saved on mg_bind().  If I want to save context data that is specific to EACH incomming connection, can I change the `user_data` pointer on each `accept` event and have multiple `user_data` pointers in each of the distinct mgConnections?

I feel that there could be room for improvement in describing the nature of `user_data` and its relationship at initialization and on separate concurrent mgConnection instances.

jpmens, 2017-06-30:
I am seeing `user_data` being set identically on all incoming connections, i.e. this appears not to be a per/connection `user_data`, and I cannot find a member which should be.

jpmens, 2017-06-30:
Thanks to https://github.com/cesanta/mongoose/issues/756 I think I've figured it out:

```c
main()
{
    struct mg_bind_opts bind_opts;

    memset(&bind_opts, 0, sizeof(bind_opts));
    bind_opts.user_data = "HELLO DOLLY";

    c = mg_bind_opt(&mgr, "9009", ev_handler, bind_opts);
...
}

static void ev_handler(struct mg_connection *nc, int ev, void *ev_data)
{
    char *doll = (char *)nc->mgr->user_data;  // HELLO DOLLY

    switch (ev) {
        case MG_EV_ACCEPT:
            nc->user_data = "connection-specific user data";
            break;
    }
...
}
```

heilkn, 2018-02-22:
Dear jpmens,
we are also using user_data to store our connection specific application state. I figured out that every new incoming connection receives the user_data of the listening connection that was returned from bg_bind. Please see also my question https://github.com/cesanta/mongoose/issues/896

Kind regards