cesanta/mongoose #822: HTTPS + ESP8266 + RTOS?

malachib, 2017-06-06:
I'm trying to utilize ESP-OPEN-RTOS with an SSL HTTP server, but encountering quite a few difficulties.

It seems that the "simplest_web_server_ssl" uses calls which eventually trickle down to mbedtls file reference API.  These API don't seem to be available.  Specifically:

1. mg_bind_opt subsequently calls
2. mbedtls_x509_crt_parse_file which depends on
3. #include <dirent.h>

Step 2 is the first failure point one hits, since by default filesystem isn't enabled.  However, when one tries to enable filesystem via `MBEDTLS_FS_IO=1` then you run into Step 3.  So far, I haven't resolved how to provide dirent.h

In Mongoose OS dirent.h is available, as evidenced by:

https://github.com/cesanta/mongoose-os/blob/master/fw/platforms/esp8266/src/esp_fs.c
https://github.com/cesanta/mongoose-os/tree/master/fw/platforms/esp8266/include

But obviously our target is mongoose, not mongoose os

I'm not sure how to proceed from here

malachib, 2017-06-06:
Somewhat related to #589

cpq, 2017-06-06:
If you want to use SSL with Mongoose, which SSL library do you intend to use?

malachib, 2017-06-07:
I plan to use mbedtls, as both Mongoose and ESP-OPEN-RTOS use it.  I'm open to something else though

anyn99, 2017-07-13:
You need a implementation for mbedtls_x509_crt_parse_file using spiffs.
I did get esp-open-rtos running with mbedtls and mongoose. Will post it here.

malachib, 2017-07-13:
Please do :)

anyn99, 2017-07-13:
I looked it up and can't really trace all the steps. The cleanest way would be to implement dirent.h for your project, but i think i just couldn't figure out how at some point.
Here is what i did: Basically you just need the functions in pkparse.c from mbedtls, and they work without dirent stuff. fopen and friends are implemented by spiffs in esp-open-rtos.
So i just made a mbedtls_fix.c or however you name it and copy everything between #if defined(MBEDTLS_FS_IO) and #endif /* MBEDTLS_FS_IO */ in pkparse.c into it.
```
/* Implementation that should never be optimized out by the compiler */
static void mbedtls_zeroize( void *v, size_t n ) {
    volatile unsigned char *p = v; while( n-- ) *p++ = 0;
}

/*
 * Load all data from a file into a given buffer.
 *
 * The file is expected to contain either PEM or DER encoded data.
 * A terminating null byte is always appended. It is included in the announced
 * length only if the data looks like it is PEM encoded.
 */
int mbedtls_pk_load_file( const char *path, unsigned char **buf, size_t *n )
{
    FILE *f;
    long size;

    if( ( f = fopen( path, "rb" ) ) == NULL )
        return( MBEDTLS_ERR_PK_FILE_IO_ERROR );

    fseek( f, 0, SEEK_END );
    if( ( size = ftell( f ) ) == -1 )
    {
        fclose( f );
        return( MBEDTLS_ERR_PK_FILE_IO_ERROR );
    }
    fseek( f, 0, SEEK_SET );

    *n = (size_t) size;

    if( *n + 1 == 0 ||
        ( *buf = mbedtls_calloc( 1, *n + 1 ) ) == NULL )
    {
        fclose( f );
        return( MBEDTLS_ERR_PK_ALLOC_FAILED );
    }

    if( fread( *buf, 1, *n, f ) != *n )
    {
        fclose( f );
        mbedtls_free( *buf );
        return( MBEDTLS_ERR_PK_FILE_IO_ERROR );
    }

    fclose( f );

    (*buf)[*n] = '\0';

    if( strstr( (const char *) *buf, "-----BEGIN " ) != NULL )
        ++*n;

    return( 0 );
}

/*
 * Load and parse a private key
 */
int mbedtls_pk_parse_keyfile( mbedtls_pk_context *ctx,
                      const char *path, const char *pwd )
{
    int ret;
    size_t n;
    unsigned char *buf;

    if( ( ret = mbedtls_pk_load_file( path, &buf, &n ) ) != 0 )
        return( ret );

    if( pwd == NULL )
        ret = mbedtls_pk_parse_key( ctx, buf, n, NULL, 0 );
    else
        ret = mbedtls_pk_parse_key( ctx, buf, n,
                (const unsigned char *) pwd, strlen( pwd ) );

    mbedtls_zeroize( buf, n );
    mbedtls_free( buf );

    return( ret );
}

/*
 * Load and parse a public key
 */
int mbedtls_pk_parse_public_keyfile( mbedtls_pk_context *ctx, const char *path )
{
    int ret;
    size_t n;
    unsigned char *buf;

    if( ( ret = mbedtls_pk_load_file( path, &buf, &n ) ) != 0 )
        return( ret );

    ret = mbedtls_pk_parse_public_key( ctx, buf, n );

    mbedtls_zeroize( buf, n );
    mbedtls_free( buf );

    return( ret );
}
``` 
A little bit hacky, but should do the trick, tell me if not.

malachib, 2017-07-13:
I'll try this out, thank you for this

cpq, 2017-07-14:
what hardware platform do you intend to use ?

cpq, 2017-07-17:
OK, 
for ESP8266, we provide mongoose-os
open-rtos is a dead end, it provides almost no RAM for the user app.

Closing this as having to real potential.

closed 2017-07-17 by cpq