cesanta/mongoose #827: Attempt to retrieve remote socket address crashes server

DoumanAsh, 2017-06-26:
I'm using following code to retrieve remote host IP address:
```c++
std::string Request::remote_ip() const {
    char buffer[40];

    mg_conn_addr_to_str(this->conn, buffer, sizeof(buffer), MG_SOCK_STRINGIFY_IP | MG_SOCK_STRINGIFY_REMOTE);

    return std::string(buffer);
}
```

Where this->conn is a `struct mg_connection*` passed to ev_handler.

DrMemory points out that there is issue when accessing pointer:
```
Error #1: UNADDRESSABLE ACCESS: reading 0xcccccd44-0xcccccd48 4 byte(s)
# 0 mg_if_get_conn_addr                         [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:2954]
# 1 mg_conn_addr_to_str                         [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:9581]
# 2 http::Request::remote_ip                    [d:\repos\cpp-code\gomibako\src\http\message.cpp:27]
# 3 route_ip                                    [d:\repos\cpp-code\gomibako\src\main.cpp:22]
# 4 handle_http_request                         [d:\repos\cpp-code\gomibako\src\http.cpp:17]
# 5 ev_handler                                  [d:\repos\cpp-code\gomibako\src\http.cpp:33]
# 6 mg_call                                     [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:2037]
# 7 mg_http_call_endpoint_handler               [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:7994]
# 8 mg_http_handler                             [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:5905]
# 9 mg_call                                     [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:2037]
#10 mg_recv_common                              [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:2512]
#11 mg_if_recv_tcp_cb                           [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:2516]
#12 mg_handle_tcp_read                          [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:3388]
#13 mg_mgr_handle_conn                          [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:3513]
#14 mg_socket_if_poll                           [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:3705]
#15 mg_mgr_poll                                 [d:\repos\cpp-code\gomibako\3pp\mongoose-6.8\mongoose.c:2231]
#16 http::Server::start                         [d:\repos\cpp-code\gomibako\src\http.cpp:61]
#17 main                                        [d:\repos\cpp-code\gomibako\src\main.cpp:32]
Note: @0:00:03.530 in thread 4388
Note: instruction: mov    0x78(%eax) -> %ecx
```

I'm using latest stable release from https://github.com/cesanta/mongoose/releases/tag/6.8

As i'm not very well familiar with mongoose, but I guess it should not at least crash server...?

cpq, 2017-06-27:
Thanks.
Could you check please what is the value of `this->conn` and `this->conn->sock` at the time of the crash please?

DoumanAsh, 2017-06-27:
@cpq `this->conn` is definitely not NULL, but I will check all other pointers that are used in `mg_if_get_conn_adr` later on.

DoumanAsh, 2017-06-27:
@cpq I'm sorry for false issue. It turns out that `this->conn` got corrupted but it wasn't nulled.
It is rather strange that Dr Memory couldn't detect it.

closed 2017-06-27 by DoumanAsh

cpq, 2017-06-28:
Great thanks for letting me know @DoumanAsh