cesanta/mongoose #834: thread read thread write

rollen1960, 2017-07-06:
I have a server that has the main loop (mg_mgr_poll / MgServer_ev_handler) on one thread and 
the send operations (mg_send) on another thread. That works but if the send operation are too fast it crash with segmentation fault in mbuf_insert function.
I suppose that it is not possible to have two different threads, one for read and one for write. Is that right ? If so how can I resolve my problem (how can I write with the same thead that read) ?

Thank you

Mario

rollen1960, 2017-07-07:
Thank you ...
Is possible to 'send' a mg_broadcast only to the specific connection that has to write ? For example is possible to write directly the socket (c->sock  f.e.  ::send(c->sock, "c", 1, 0) ) so the manager calls event handler with MG_EV_POOL only for that connection ?

CfirTsabari, 2017-07-07:
you are welcome,
one option you can do is:
1)use different callback in mg_binding and mg_connection
2)use the data you can send to mg_broadcast to uniquely identify the mg_connection you want to send to the data.
3)uniquely identify the mg_connection using maybe the sock member but even better user_data.

but even better option(especially if you are using c++):
create for each connection a list(queue) and push messages to there,and when the mg_broadcast arrive each connection should just empty it's queue by sending it.
 
if you have any more question about how to do that ,contact me again.

rollen1960, 2017-07-12:
All right but the POLL event come for all connection and come with or without mg_broadcast.
Thank you

closed 2017-07-12 by rollen1960