cesanta/mongoose #835: http pipelining

CfirTsabari, 2017-07-07:
hello,
i am trying to create a http client that do http pipelining so i won't have to closed the connection.
but this cause an assertion at the code specifically [here](https://github.com/cesanta/mongoose/blob/master/mongoose.c#L5456).
basiclly what i do is `mg_connect_opt()` and then `mg_set_protocol_http_websocket() `after this i send one request using `mg_printf()` and after 5 seconds i send another request using `mg_printf() `the assertion received for the second request.
i think i know how to fix it but i am not sure, is this by design or a bug?

cpq, 2017-07-17:
Please publish your code - we need a reliable way to reproduce.

CfirTsabari, 2017-07-17:
reproduce using this code [here](https://gist.github.com/CfirTsabari/bd120a1d3abbabd307e31eb75fda493c)

ruslanvaliullin, 2017-07-17:
ok. thx

ruslanvaliullin, 2017-08-18:
@CfirTsabari, sorry for the late reply. This issue is not reproduced for me. I make the http client and sent a request with `mg_print` every second for the same connection. There is no assertion. Yes, each request is added to the buffer, but this is the expected behaviour. Please clarify.

CfirTsabari, 2017-08-21:
hello,
i am not sure why this is not reproducing for you, i just check and it's still reproducing for me.
it might be OS related but it's unlikely, i am using windows 7 x64 and VS 2013.
i fixed this on my code by add this code of line:
`	  
pd->chunk.body_len = 0;
`
right after the line number 5498 inside the if:
`if (zero_chunk_received) `

ruslanvaliullin, 2017-08-21:
How build your [code](https://gist.github.com/CfirTsabari/bd120a1d3abbabd307e31eb75fda493c)? Could you please give your VS project?

CfirTsabari, 2017-08-21:
i will try to do it shortly

CfirTsabari, 2017-08-25:
hello @ruslanvaliullin  ,
sorry for the delay,
here is the an example i made for testing this issue  in VS 2013.
When  REQUEST_NUM is set to 1 everything work ,but when setting REQUEST_NUMto anything greater than 1 an assertion occurred. 

[Test HTTP PIPE line.zip](https://github.com/cesanta/mongoose/files/1252482/Test.HTTP.PIPE.line.zip)

cpq, 2020-12-14:
Thank you.

The https://github.com/cesanta/mongoose/blob/master/test/unit_test.c has `test_pipeline()` function that passes for Win32.
Note that pipelining works only when responses have Content-Length set, or use chunked encoding.

If pipelining is broken for you for some reason, feel free to send a PR against the unit test.

closed 2020-12-14 by cpq