cesanta/mongoose #836: mqtt_client Do not support the Apollo broker

skyformat99, 2017-07-11:
1 Use Apollo to build the broker:
http://activemq.apache.org/apollo/

2 compile mqtt_client ,then run it
       $ ./mqtt_client -h 192.168.2.55:1883 -p 61613 -u admin -p password

3  use other client, publish message, 
    The mqtt_client cannot receive messages .

if use other broker (ex: http://emqtt.io/) ,  all is ok.

ruslanvaliullin, 2017-08-19:
@skyformat99, sorry for the late reply. I used mosquitto to publish message and `mqtt_client` works fine.

closed 2017-08-19 by ruslanvaliullin

skyformat99, 2017-08-19:
It's no use having your head in the sand . You should try to solve the problem.

ruslanvaliullin, 2017-08-20:
Are you sure that there is a problem with mqtt_client? This problem is not reproduced for me. What kind of response did you expect to hear by asking such a question? Please try clarify what does not work exactly, give at least a log and a config so that we can see what you are trying to do. And we'll try to help you.

skyformat99, 2017-08-20:
ref
https://github.com/romuloceccon/lightmqtt/issues/1
[ the issue you experienced was due to an incorrect implementation of the MQTT 3.1.1 spec regarding the handling of QoS 0 messages. It's now fixed in commit b9da410.

Incidentally, the bug became evident because the Apollo broker also seems to not follow the specs correctly: even when receiving QoS 2 messages it forwards them using QoS 0.]