cesanta/mongoose #841: Significant Bug when dealing with QoS  publish messages which qos is 1

zhaojh329, 2017-07-17:
case MG_MQTT_CMD_PUBLISH: {
      if (MG_MQTT_GET_QOS(header) > 0) {
        mm->message_id = getu16(p);
        p += 2;
      }
      p = scanto(p, &mm->topic);

      mm->payload.p = p;
      mm->payload.len = end - p;
      break;
    }

The correct code is as follows:

case MG_MQTT_CMD_PUBLISH: {
      p = scanto(p, &mm->topic);
	  if (MG_MQTT_GET_QOS(header) > 0) {
        mm->message_id = getu16(p);
        p += 2;
      }
      mm->payload.p = p;
      mm->payload.len = end - p;
      break;
    }
see [evmongoose](https://github.com/zhaojh329/evmongoose/commit/889cf5b1de7e10cc94355cdf227b5c1f15af7e19)

cpq, 2017-07-17:
Please stop referencing the irrelevant parts of your project.
Continuing doing so will result in a ban.

cpq, 2017-07-17:
If you want to contribute, sign the CLA at https://docs.cesanta.com/contributors_la.shtml
and send a PR.

closed 2018-02-01 by dimonomid