cesanta/mongoose #844: Hang on connect()

timredfern, 2017-07-26:
We're seeing an erratic issue that makes the mongoose webserver hang.

Platform is cc3200 - mongoose 6.7 - cc3200 SDK 1.2.0

The issue is affected by whether the firmware is flashed, or running in the debugger. Code that works in the debugger will hang on first connect() when flashed. We are using Ti's OAD bootloader, so it seems that something relies on memory being zeroed and when it isn't, the code can hang.

The hang occurs when mongoose tries to access the filing system. The workaround is to disable mongoose from trying to access resolv.conf and /etc/hosts, by returning -1 in the functions `mg_get_ip_address_of_nameserver` and `mg_resolve_from_hosts_file`: 

on line 10669

`FILE *fp;
  char line[512];

  return -1;
  if ((fp = mg_fopen("/etc/resolv.conf", "r")) == NULL) {`

and on line 10700

`int len = 0;

  return -1;
  if ((fp = mg_fopen("/etc/hosts", "r")) == NULL)`

I can see that mg_fopen is just a wrapper to fopen - it's unclear why this issue occurs when mongoose tries to read these files but never during other filesystem accesses.

cpq, 2017-07-27:
Already fixed in the dev branch.
You can wait for the next release, or cherry pick this commit:

https://github.com/cesanta/mongoose/commit/f868fc861b68bfdd59cc193b893966061c8aa5a3

timredfern, 2017-07-27:
OK thanks, I see the fix is the same, to disable the files from being used.

Are you seeing the hanging behaviour anywhere else? Seems to be a cc3200 SDK issue more than mongoose?

cpq, 2017-07-28:
That is the CC3200 issue, correct

cpq, 2017-07-28:
Btw, do you guys plan to switch to CC3220 ?
It is a "grown up" version of CC3200, which includes serious improvements, and has a crypto chip onboard for securing the TLS credentials.

timredfern, 2017-07-28:
It's on the cards, yes, not for the first iteration though.

closed 2018-02-01 by dimonomid