cesanta/mongoose #845: examples connected_device_4 segmentation fault on Raspberry Pi

ozlb, 2017-08-15:
if in server.c you add at the begin 
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

pi@raspberrypi ~/projects/c/mongoose-master/examples/connected_device_4 $ ./server
Starting web server on port 8000
Segmentation fault
pi@raspberrypi ~/projects/c/mongoose-master/examples/connected_device_4 $ gdb server
GNU gdb (GDB) 7.4.1-debian
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/pi/projects/c/mongoose-master/examples/connected_device_4/server...done.
(gdb) run
Starting program: /home/pi/projects/c/mongoose-master/examples/connected_device_4/server 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Starting web server on port 8000

Program received signal SIGSEGV, Segmentation fault.
0x0001414c in mg_socket_if_poll (iface=0x30018, timeout_ms=1000) at ../../mongoose.c:3602
3602	    tmp = nc->next;

Attila888, 2018-03-14:
Same problem with Wheezy and Stretch on raspberry PI2 and mongoose 6.11 and 6.09:
th_WWWsrv(): starting web server on port 8080

Thread 9 "Prg_Look@Home20" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7341e450 (LWP 2773)]
0x0001d550 in mg_socket_if_poll (iface=0x71a00480, timeout_ms=1000)
    at www-src/mongoose.c:3783
3783        tmp = nc->next;

cpq, 2020-12-14:
Closing this.

Please use a `complete` example, which is a dashboard for connected device.

closed 2020-12-14 by cpq