cesanta/mongoose #854: Mongoose is blocking other requests and processing them one by one

cemiltokatli, 2017-09-13:
I tried to run "restful_server" example under the "examples". ev_handler function processes all requests one by one so a request has to wait until the prior request has been completed.

```
static void handle_sum_call(struct mg_connection *nc, struct http_message *hm) {
  char n1[100], n2[100];
  double result;

  /* Get form variables */
  mg_get_http_var(&hm->body, "n1", n1, sizeof(n1));
  mg_get_http_var(&hm->body, "n2", n2, sizeof(n2));
  
  	int n2s = (int) strtod(n2, NULL);
	if(n2s == 100){
		sleep(6000);
	}
	
  /* Send headers */
  mg_printf(nc, "%s", "HTTP/1.1 200 OK\r\nTransfer-Encoding: chunked\r\n\r\n");

  /* Compute the result and send it back as a JSON object */
  result = strtod(n1, NULL) + strtod(n2, NULL);
  mg_printf_http_chunk(nc, "{ \"result\": %lf }", result);
  mg_send_http_chunk(nc, "", 0); /* Send empty chunk, the end of response */
}
```

I've also tried this by adding a sleep function into handle_sum_call. I was planning to use mongoose to create a simple web application that will be used by a 10-20 users (not too much). It says that it is asynchronous and non-blocking? Am I doing something wrong?

cpq, 2017-09-13:
Yes you're doing something wrong.
Mongoose event handler must not block. `sleep()` blocks. A blocking call blocks everything.

closed 2017-09-13 by cpq

cemiltokatli, 2017-09-13:
Yeah I know but think about a heavy progress like I/O or a calculation? It will also block everything.

cpq, 2017-09-13:
That should go to a different thread, and communicate a result via `mg_broadcast()`

cpq, 2017-09-13:
See https://github.com/cesanta/mongoose/tree/dev/examples/multithreaded

cemiltokatli, 2017-09-13:
That is the answer I was looking for. Thank you.