cesanta/mongoose #857: Corruption of data when passing user_data directly through mg_bind

Silverspur, 2017-09-16:
I have the following code, with `Base` and `Child` being two classes, `Child` inheriting from `Base`.

    _connection =
    mg_bind( mgr, "8000", handle_event, static_cast<Base*>(myChild) );

And the callback:

    void handle_event( mg_connection* connection, int event, void* event_data, void* user_data ){
        // this calls Base::method instead of Child::method
        (static_cast<Base*>user_data)->method();
    }

It looks like the vtable has disapeared in the process...
The issue is solved when using mg_bind with only 3 arguments (disable MG_ENABLE_CALLBACK_USERDATA) and passing directly the user data to the connection after the call to `mg_bind`:

    _connection->user_data = static_cast<Base*>(myChild);

and using the connection to retrieve the user data in the callback:

    (static_cast<Base*>(connection->user_data))->method();

This issue looks very much like #756 but in monothread.

milmarqtan, 2019-02-07:
__connection =
mg_bind( mgr, "8000", handle_event, static_cast<Base*>(myChild) );_

Have you tried casting myChild to a void * instead of Base *?

cpq, 2020-12-14:
The C++ class members, if used as callbacks (mongoose event handlers), must be declared as static.

closed 2020-12-14 by cpq