cesanta/mongoose #864: error: use of undeclared identifier 'fchmod'

zhou0, 2017-10-04:
gmake[1]: Entering directory '/usr/home/lzhou/github.com/cesanta/mongoose/archive/mongoose-6.0/examples/api_server'
cc api_server.c sqlite3.c db_plugin_sqlite.c ../../mongoose.c -o api_server -W -Wall -pthread 
sqlite3.c:15249:49: warning: unused parameter 'id' [-Wunused-parameter]
SQLITE_PRIVATE int sqlite3OsFetch(sqlite3_file *id, i64 iOff, int iAmt...
                                                ^
sqlite3.c:15249:57: warning: unused parameter 'iOff' [-Wunused-parameter]
SQLITE_PRIVATE int sqlite3OsFetch(sqlite3_file *id, i64 iOff, int iAmt...
                                                        ^
sqlite3.c:15249:67: warning: unused parameter 'iAmt' [-Wunused-parameter]
  ...int sqlite3OsFetch(sqlite3_file *id, i64 iOff, int iAmt, void **pp){
                                                        ^
sqlite3.c:15253:51: warning: unused parameter 'id' [-Wunused-parameter]
SQLITE_PRIVATE int sqlite3OsUnfetch(sqlite3_file *id, i64 iOff, void *p){
                                                  ^
sqlite3.c:15253:59: warning: unused parameter 'iOff' [-Wunused-parameter]
SQLITE_PRIVATE int sqlite3OsUnfetch(sqlite3_file *id, i64 iOff, void *p){
                                                          ^
sqlite3.c:15253:71: warning: unused parameter 'p' [-Wunused-parameter]
SQLITE_PRIVATE int sqlite3OsUnfetch(sqlite3_file *id, i64 iOff, void *p){
                                                                      ^
sqlite3.c:23527:42: error: use of undeclared identifier 'fchmod'
  { "fchmod",       (sqlite3_syscall_ptr)fchmod,     0  },
                                         ^
sqlite3.c:23587:22: error: invalid application of 'sizeof' to an incomplete
      type 'struct unix_syscall []'
    for(i=0; i<sizeof(aSyscall)/sizeof(aSyscall[0]); i++){
                     ^~~~~~~~~~
sqlite3.c:23596:22: error: invalid application of 'sizeof' to an incomplete
      type 'struct unix_syscall []'
    for(i=0; i<sizeof(aSyscall)/sizeof(aSyscall[0]); i++){
                     ^~~~~~~~~~
sqlite3.c:23623:20: error: invalid application of 'sizeof' to an incomplete
      type 'struct unix_syscall []'
  for(i=0; i<sizeof(aSyscall)/sizeof(aSyscall[0]); i++){
                   ^~~~~~~~~~
sqlite3.c:23640:16: error: invalid application of 'sizeof' to an incomplete
      type 'struct unix_syscall []'
    for(i=0; i<ArraySize(aSyscall)-1; i++){
               ^~~~~~~~~~~~~~~~~~~
sqlite3.c:8375:38: note: expanded from macro 'ArraySize'
#define ArraySize(X)    ((int)(sizeof(X)/sizeof(X[0])))
                                     ^~~
sqlite3.c:23644:14: error: invalid application of 'sizeof' to an incomplete
      type 'struct unix_syscall []'
  for(i++; i<ArraySize(aSyscall); i++){
             ^~~~~~~~~~~~~~~~~~~
sqlite3.c:8375:38: note: expanded from macro 'ArraySize'
#define ArraySize(X)    ((int)(sizeof(X)/sizeof(X[0])))
                                     ^~~
sqlite3.c:27694:37: warning: unused parameter 'pFd' [-Wunused-parameter]
static void unixUnmapfile(unixFile *pFd){
                                    ^
sqlite3.c:27824:34: warning: unused parameter 'pFd' [-Wunused-parameter]
static int unixMapfile(unixFile *pFd, i64 nByte){
                                 ^
sqlite3.c:27824:43: warning: unused parameter 'nByte' [-Wunused-parameter]
static int unixMapfile(unixFile *pFd, i64 nByte){
                                          ^
sqlite3.c:27868:36: warning: unused parameter 'fd' [-Wunused-parameter]
static int unixFetch(sqlite3_file *fd, i64 iOff, int nAmt, void **pp){
                                   ^
sqlite3.c:27868:44: warning: unused parameter 'iOff' [-Wunused-parameter]
static int unixFetch(sqlite3_file *fd, i64 iOff, int nAmt, void **pp){
                                           ^
sqlite3.c:27868:54: warning: unused parameter 'nAmt' [-Wunused-parameter]
static int unixFetch(sqlite3_file *fd, i64 iOff, int nAmt, void **pp){
                                                     ^
sqlite3.c:41710:37: warning: unused parameter 'pPager' [-Wunused-parameter]
static void pagerFixMaplimit(Pager *pPager){
                                    ^
13 warnings and 6 errors generated.
gmake[1]: *** [Makefile:16: api_server] Error 1
gmake[1]: Leaving directory '/usr/home/lzhou/github.com/cesanta/mongoose/archive/mongoose-6.0/examples/api_server'
gmake: *** [Makefile:12: api_server/] Error 2

cpq, 2017-10-04:
Please use the latest version

closed 2017-10-04 by cpq

zhou0, 2017-10-04:
version 6.9 has the same issue.

cpq, 2017-10-04:
This is not an issue with Mongoose.
This call is made by the sqlite3 .