cesanta/mongoose #870: duped sockets not be closed properly?

air33, 2017-11-08:
Occasionally I've watched  about long-lived large number of open sockets  with the use of mongoose lib  under very low load.
So I seek into the source code , find the following lines(develop branch) in function `mg_socket_if_poll` :

```
#ifdef __unix__
      /* A hack to make sure all our file descriptos fit into FD_SETSIZE. */
      if (nc->sock >= (sock_t) FD_SETSIZE && try_dup) {
        int new_sock = dup(nc->sock);
        if (new_sock >= 0 && new_sock < (sock_t) FD_SETSIZE) {
          closesocket(nc->sock);
          DBG(("new sock %d -> %d", nc->sock, new_sock));
          nc->sock = new_sock;
        } else {
          try_dup = 0;
        }
      }
#endif
```

When  the condition `new_sock >= 0 && new_sock >= (sock_t) FD_SETSIZE` is true, the duped socket will  not be closed.

Is it a leak?

dimonomid, 2018-01-30:
But instead, the old socket is closed then, and the newly open socked is saved as `nc->sock`. I don't see a leak here. Am I missing something?

dimonomid, 2018-01-30:
Ah right, I misread. You're right, there's a leak, will be fixed soon. Thanks for reporting!

closed 2018-01-30 by dimonomid