cesanta/mongoose #882: esp32_gpio.c uses wrong register to clear interrupt for GPIO pin >=32

robertfarnum, 2017-12-27:
Diff for changes required to resolve issue:

gpio.c
index 9ed00c96..9fea466f 100644
--- a/fw/platforms/esp32/src/esp32_gpio.c
+++ b/fw/platforms/esp32/src/esp32_gpio.c
@@ -49,13 +49,14 @@ IRAM void mgos_gpio_hal_int_clr(int pin) {
   uint32_t reg = GPIO_STATUS_W1TC_REG;
   if (pin >= 32) {
     pin -= 32;
-    reg += 4; /* GPIO_STATUS_W1TC -> GPIO_STATUS1_W1TC */
+    reg = GPIO_STATUS1_W1TC_REG;
   }
   WRITE_PERI_REG(reg, (1 << pin));
 }
 
 IRAM void mgos_gpio_hal_int_done(int pin) {
   mgos_gpio_hal_int_clr(pin);
+  printf("#### mgos_gpio_hal_int_done(%d): int_ena = %d\n", pin, s_int_ena[pin]);
   GPIO.pin[pin].int_ena = s_int_ena[pin];
 }

Technical Reference for ESP32 GPIO Registers:

GPIO_STATUS_W1TS_REG GPIO 0-31 interrupt status register_W1TS 0x3FF44048 WO
GPIO_STATUS_W1TC_REG GPIO 0-31 interrupt status register_W1TC 0x3FF4404C WO
GPIO_STATUS1_REG GPIO 32-39 interrupt status register1 0x3FF44050 R/W
GPIO_STATUS1_W1TS_REG GPIO 32-39 interrupt status bit set register 0x3FF44054 WO
GPIO_STATUS1_W1TC_REG GPIO 32-39 interrupt status bit clear register 0x3FF44058 WO

rojer, 2017-12-27:
fix sent, thank you!

rojer, 2017-12-27:
next time use https://github.com/cesanta/mongoose-os repo though

closed 2017-12-27 by rojer