cesanta/mongoose #883: user_data stolen from users by MQTT broker

schmurfy, 2018-01-03:
I tried to play with the mqtt broker support but I am quite puzzled by how bad it fits with the other pieces, the first thing that strikes me is why on earth does the broker uses `user_data` to store its own internal structure ? By doing this its means that `user_data` is no longer my own data...
The problem is that i already use `user_data` to store my own internal state :(

I also don't quite get why the broker structure has to be be initialized in user code instead of being completely handled inside the mongoose code.

Can something clarify why it works like that ? is that something you would be open to change ? A simple fix for my first issue would be to add a second pointer for the broker structure so that `user_data` is left to users.
I can make the change and submit a PR but only if there is a chance it will be merged.

cpq, 2018-01-04:
We'd be open to merge such a change - could you briefly describe it before jumping to PR please ?

schmurfy, 2018-01-06:
One straightforward option would be to add an  `internal_data` (or maybe `proto_data2` ?) field and use that for the mqtt broker object:

```c
struct mg_connection {
  // ...
  void *user_data;            /* User-specific data */
  void *internal_data;
}
```

I have been working on a binding for mruby for a while now and so far everything fits perfectly, I really like the overall design of mongoose and I was quite puzzled when tried to integrate the mqtt broker which deliberatly use user_data as it was its own :)

If the size this adds to the mg_connection is of any concern this field could be only delcared when `MG_ENABLE_MQTT_BROKER` is set, that way it would not make any difference when not used.
What do you think ?

cpq, 2018-01-08:
We'd be reluctant to add more fields to the struct.
Would existing fields like `priv_2` be enough?

schmurfy, 2018-01-09:
Maybe, honestly I have no idea what priv_2 is used for, from a quick search through the code it seems to be used for dns resolution by mg_connect, if that is the only usecase I suppose mqtt broker could use it.
I don't really mind where the mqtt broker is stored as long as it is not in `user_data`

schmurfy, 2018-01-15:
I opened a pull requests with the changes.

schmurfy, 2018-01-30:
@cpq did you had any chance to check the pr ? I did some tests before releasing it to make sure mqtt client and server still works and they do.

closed 2018-01-30 by rojer