cesanta/mongoose #887: example web server ssl dont compile

antoniosap, 2018-01-17:
`cc http_client.c ../../mongoose.c -o http_client -g -W -Wall -Werror -I../.. -Wno-unused-function   -pthread -DMG_ENABLE_SSL -lssl -lcrypto
/tmp/cc0eZNsV.o: nella funzione "mg_ssl_if_init":
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4540: riferimento non definito a "OPENSSL_init_ssl"
/tmp/cc0eZNsV.o: nella funzione "mg_ssl_if_conn_init":
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4579: riferimento non definito a "TLS_server_method"
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4581: riferimento non definito a "TLS_client_method"
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4590: riferimento non definito a "SSL_CTX_set_options"
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4591: riferimento non definito a "SSL_CTX_set_options"
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4592: riferimento non definito a "SSL_CTX_set_options"
/tmp/cc0eZNsV.o: nella funzione "mg_use_cert":
/home/antonio/esp32/mongoose/examples/http_client/../../mongoose.c:4811: riferimento non definito a "SSL_CTX_set_options"
collect2: error: ld returned 1 exit status
`

dimonomid, 2018-01-29:
I was thinking that by "example web server ssl" you refer to https://github.com/cesanta/mongoose/tree/master/examples/simplest_web_server_ssl , but it looks like you compile something else. Please clarify.

antoniosap, 2018-01-29:
on esp32 xtensa compiler

cpq, 2020-12-17:
Closing this.
Please use the latest version.

closed 2020-12-17 by cpq