cesanta/mongoose #888: Is it safe to save a mg_connection pointer, and later try to find it by comparing the nc pointer during an MG_POLL event?

Eloquence4, 2018-01-18:
My application is running on 2 threads. One of the threads runs the mongoose event loop. Sometimes i need to introduce blocking functions, and for that i communicate with the second thread to do something, meanwhile saving the pointer for the mg_connection.

After the second thread is done, i put my data and mg_connection pointer in a structure, and i mg_broadcast the pointer to that structure. After that my other mongoose event function is called. Is it safe to try to identify the connection i saved by comparing the pointers? I imagine the mg_connection structure does not get recreated anywhere, so it should be safe, but i want to make sure.

Also, i am aware of mg_connection->user_data, but i am using that for something else. I could theoretically use that for 2 things, but i believe the current solution would be simpler and easier if it's valid.

cpq, 2018-01-18:
connection pointers must not be shared between threads, regardless how do you pass them over.

the only mechanism to communicate to mongoose thread is via mg_broadcast, passing data that must NOT include any mongoose runtime - because any connection can go in any moment.

closed 2018-01-18 by cpq

Eloquence4, 2018-01-18:
Ya, i got that part, i use mg_broadcast for that. I was just asking if there was any way to identify a connection other than the user_data field. I was not going to use the connection pointer for anything other than identify the connection when i'm back in the mongoose thread.

cpq, 2018-01-22:
No you don't get it. ` i put my data and mg_connection pointer in a structure, and i mg_broadcast the pointer to that structure.` - you're passing a pointer to another thread.

Eloquence4, 2018-01-22:
Yea but i don't do anything with the pointer. I can always save random data in other threads, doesn't mean i'll do anything with it. I'm pretty sure comparing pointers is a thread safe operation, do correct me if i'm wrong.

cpq, 2018-01-22:
If you don't do anything with the pointer, that's fine. Why you're passing it over then? Mongoose thread can close the connection and use the same pointer for another one. Don't identify connections by their pointers - use some sort of auto-incrementing IDs or something like that.

Eloquence4, 2018-01-22:
I wanna identify them with the pointers because it would just make my life a lot easier, since i already use the user_data field. I would have to think of another way of solving my problem if it's safe, probably IDs as you recommended. Thank you.