cesanta/mongoose #890: Many instantaneous messages

carlkabbe1, 2018-01-29:
I have an industrial process that requires sending many state messages through a web socket to some javascript in the browser.  An executable running on the Windows PC attached through some electronics to the valves, pumps and other assorted hardware gathers state information and constantly returns it to the web app running in the browser.  The broadcast function from the example chat program worked well when the process was do some quick function and return the results to the web app but does not work well for long running functions that contain sleep or other blocking actions.  The solution appears to be using mg_broadcast on a separate thread.  I have tried using the example from here https://github.com/cesanta/mongoose/blob/dev/examples/multithreaded/multithreaded.c but I have two issues that I don't seem to be able to resolve:

1.  mg_socketpair fails on my machine (Windows 10, Visual Studio 2017, new machine, new install) at the first step if ((sock = socket(AF_INET, sock_type, 0)) == INVALID_SOCKET).  If I place the call to mg_socketpair after creating the connection, i.e., mg_mgr_init, mg_bind, etc., mg_socketpair succeeds but the test in worker_thread_proc 'if (read(sock[1], &req, sizeof(req)) < 0)' fails because, I think, sock[1], contains an invalid value.  If I eliminate the mg_socketpair call in main, then 'if (read(sock[1], &req, sizeof(req)) < 0)' always says, I think, there is nothing to do on that sock and never does anything (I am a little confused here.)

2.  I read this post:
Thanks for elaborating on this. As I said, there multiple ways to do it. One way is to keep serving core single-threaded and use mg_broadcast() API function. Here is a signature:

/*
 * Passes a message of a given length to all connections.
 *
 * Must be called from a thread that does NOT call `mg_mgr_poll()`.
 * Note that `mg_broadcast()` is the only function
 * that can be, and must be, called from a different (non-IO) thread.
 *
 * `func` callback function will be called by the IO thread for each
 * connection. When called, the event will be `MG_EV_POLL`, and a message will
 * be passed as the `ev_data` pointer. Maximum message size is capped
 * by `MG_CTL_MSG_MESSAGE_SIZE` which is set to 8192 bytes.
 */
void mg_broadcast(struct mg_mgr *, mg_event_handler_t func, void *, size_t);
The sequence of actions is this:

When /compute POST request is received, pass computation parameters to another thread. Pass some ID to identify current connection.
When computation is finished, call mg_broadcast. Pass that ID and computation -results in the message.
In the callback function, check the current connection by comparing the ID. If you're on a connection that is waiting for the result, fetch the result from the message and send the reply back to the client.

by Sergey Dublin, Ireland
September 2016

but still am uncertain how this actually works, i.e., do I a) create a thread using mg_start_thread, send the message, and kill/stop the thread (to be repeated for each message) or b) do I start the thread  and have it do every message (not certain how to do this.)

I understand this is a big request... thanks for any suggestions or other help.

Carl

moranxy, 2020-03-30:
I have same problem。When I use the  "WSAStartup()" before "sock = socket(AF_INET, sock_type, 0)) == INVALID_SOCKET"," _VALIDATE_CLEAR_OSSERR_RETURN(fh >= 0 && (unsigned)fh < (unsigned)_nhandle, EBADF, -1)" will break, can't read anymore,even i have not close the sock.

moranxy, 2020-03-30:
do you solve that problem?

cpq, 2020-12-14:
The `mg_broadcast` API is gone since the recent 7.0 release.
Please see https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded example

closed 2020-12-14 by cpq