cesanta/mongoose #893: How to use openssl with mongoose to enable ssl?

xyuan240, 2018-01-31:
Hi, I am trying to enable ssl for moongoose webserver, and I have successfully compiled openssl (ver1.0.2) configured as VC-WIN64A(MSVC2008 ).

When I tried to include lib files (libssl.lib and libcrypto.lib as shown in your ssl example). These two files are not generated anymore in openssl v1.0.2.  I wonder if you can provide some guidance on how to use openssl with mongoose? Thanks in advance.

Regards

CfirTsabari, 2018-11-08:
Hi i am using open ssl 1.1.0f and it's work just find.
The libraries names can be a little different for example in my debug setup i use libsslMDd.lib  and libcryptoMDd.lib that because i use dlls,
what libraries you do see in the libs folder of the open ssl directory ?

cpq, 2020-12-14:
Closing this.

Please see https://github.com/cesanta/mongoose/tree/master/examples/http-client for an example.

closed 2020-12-14 by cpq