cesanta/mongoose #896: Do not Assign user_data to New Connections

heilkn, 2018-02-13:
Hello,
we've built a C++ implementation on top of the mongoose library. We are storing a pointer to a ``class Connection`` in the user_data of every mg_connection. However, when a new mg_connection is created by mongoose we have observed that it carries the user_data of the listening connection. This is due to the statement
```nc->user_data = lc->user_data;```
in the method ``mg_if_accept_new_conn``.
We'd like to suggest changing that to 
```nc->user_data = nullptr;```

Do you we mish to create a pullrequest?

Kind regards

Konstantin

PuKathy, 2018-11-07:
How about set nc->user_data = NULL at server handler's MG_EV_ACCEPT event ？
If you do not use the user_data, I think "nc->user_data = lc->user_data" will not affect.

heilkn, 2018-11-07:
Hello,
thank you for your response. We use the user data, and we had to deal with the problem, that the poll event for a new connection was triggered before we could assign it a new user_data. The event handler was then working incorrectly, because the user_data attached to that new connection was not correct, because it was copied from the listening connection.

Of course, we could implement a workaround but we have created an internal fork currently.
Why do you think the current implementation in mg_if_accept_new_conn is correct?

Kind regards

Konstantin

cpq, 2020-12-14:
Closing this.

Feel free to reopen - if you do, please share a minimal reproducible example of your code.

closed 2020-12-14 by cpq