cesanta/mongoose #901: Windows version is flagged by 24 antivirus....

propergol, 2018-02-19:
Something in the code of the Windows install file is flagged by 24 AV on Virustotal : 
https://www.virustotal.com/fr/file/6d0931422494e66aefc96ac4d76808ce6da274c120350bad86ff4df6f9e64623/analysis/1519051920/

I did test the trial install file and bought the pro version : same issue.

Can you please comment?

Seikilos, 2018-03-04:
Same happened here. The file differs from v6.5 significantly as it is only ~144kb in size, where 6.5 was ~532kb

group131, 2018-04-03:
Any fix?

icie-y, 2018-05-15:
+1 because the file mongoose-free-6.9.exe was marked as malicious by the security team in my university

group131, 2018-05-15:
In my case I ended up using version 6.5. So my question is: Why is version 6.5 not flagged? What's the difference?

rojer, 2018-05-15:
nothing substantial. we use the same toolchain to build binaries, nothing really changed since 6.5 (apart from features and bugfixes).

cpq, 2018-05-17:
@group131 , you've been told that mongoose.exe has no malicuous code inside.

If you're paranoid, don't use the binary, it is that simple.

closed 2018-05-17 by cpq

AdminUser0, 2020-06-04:
> 
> 
> @group131 , you've been told that mongoose.exe has no malicuous code inside.
> 
> If you're paranoid, don't use the binary, it is that simple.

nice answer!