cesanta/mongoose #903: Performance of mg_send_websocket_frame

xyuan240, 2018-02-26:
When I use function mg_send_websocket_frame to send a string message (length is 277)  to client, I notice that there is an apparent latency between the end of function call mg_send_websocket_frame to the point that the client browser receives the message. The elapsed time is around 950ms, given that server and client are both located on the same machine, this time is too long for a string with length of 200~.

I tried some websocket client/server implementation in node.js (nodejs-websocket), the latency time is around 20-50ms for the same message to reach the same client.

I wonder if you are aware of this issue and are there any ways to improve the performance? Many thanks 

With regards
xyuan

PuKathy, 2018-11-07:
mg_send_websocket_frame only put data in buffer, it will actually send out when mg_mgr_poll().
Maybe you can adjust the second parameter of function mg_mgr_poll()

cpq, 2020-12-17:
Closing this.
Please use the latest version.

closed 2020-12-17 by cpq

volanin, 2021-08-02:
I was having this same problem and arrived here via Google. I'm posting this to clarify the solution to other people that might need it. As @pukaifei said, **mg_send_websocket_frame()** in V6 and **mg_ws_send()** in V7 only put the data in the send buffer. It's just when you explicitly call **mg_mgr_poll()** that this data is sent.

This is not very clear in the provided mongoose examples, because they use the following code snippet, flushing the messages only once per second, without much explanation:

```
for (;;) mg_mgr_poll(&mgr, 1000);             // Infinite event loop
```

In order to immediately send the message, just do this:

```
// In V6:
mg_send_websocket_frame(connection, WEBSOCKET_OP_TEXT, data, size);
mg_mgr_poll(&mgr, 0);

// In V7:
mg_ws_send (connection, data, size, WEBSOCKET_OP_TEXT);
mg_mgr_poll(&mgr, 0);
```

cpq, 2021-08-03:
Thank you @volanin 

The correct way is coded in https://github.com/cesanta/mongoose/tree/master/examples/websocket-client

Note that mg_mgr_poll() returns immediately if there is something to process. It waits only if there is nothing to process. And a single call may not send the whole buffer - it could send only a part of it. So a correct way is to send, then loop until everything is sent.

If there are many connections, just use a single loop for all of them. There should not be any delays.

volanin, 2021-08-04:
[mg_mgr_wake.patch.gz](https://github.com/cesanta/mongoose/files/6927726/mg_mgr_wake.patch.gz)

Proposed patch for **mg_mgr_wake()** using **mg_socketpair()**
It needs MG_ENABLE_SOCKETPAIR to be true. If not, the **mg_mgr_wake()** simply does nothing.

volanin, 2021-08-05:
I opted to open a new issue ticket for this.
Please go to #1332