cesanta/mongoose #905: Memory corrupted when tries to connect to an unreachable address

paladin-t, 2018-03-13:
Environment: Windows 7 64bit, Visual C++ 2015 (Debug)

I'm just messing with the `udp_client.c`, when I pass `127.0.0.1:17000 hello` to the program (the address is unreachable), it invokes the callback with freed memory where has been tagged `0xdddddddd...`.

[image]

[image]

It tries to free that chunk of memory which already has been freed, when I modify the source to only polling once.

[image]

Thanks.

cpq, 2020-12-14:
Could you reproduce this with the latest code ?

paladin-t, 2020-12-15:
The corruption is gone. It's infinite wait now, I suppose it is expected behaviour.

[image]

cpq, 2020-12-15:
Excellent, thank you for investing time into it.

Closing this issue.

closed 2020-12-15 by cpq