cesanta/mongoose #907: Plain UDP does not preserve packet borders

numo68, 2018-03-23:
This issue is based just on looking at the 6.11 code while assessing of whether it is suitable for implementing a general UDP communication. I cannot exclude that I am overlooking something but I'll document the issue nevertheless.

One of the primary features of UDP is that it preserves the message borders - if I send a 1024 bytes packet, a 16384 one and a 384 one using three sendto calls, the other side gets those exact data in three subsequent recvfrom calls.

If I look into the mongoose code there is just one mbuf for the connection to send the data from. Two `mg_send`s without the underlying `sendto` (for whatever reason) will result in one packet containing data from both sends.

In the receiving direction the `mg_handle_udp_read` calls `mg_recvfrom` that uses a fixed buffer of a 1500 size and appends the data to a receiving mbuf. The code is completely unable to accept packets larger than a compile-time defined size, as the `recvfrom` on a datagram socket will silently truncate the excess data.

That means that the `MG_F_UDP` code is basically unusable for what the UDP communication is usually being used for, i.e. being able to send datagrams that are unreliable and might be out of order or duplicated, but a datagram is either completely received or not at all. If this is indeed the case I'd recommend to either fix it (e.g. by using linked mbufs, each containing one and only one packet, and a configurable buffer at the receiving side), or to remove the functionality completely, as it will only work for packets <= 1500 bytes being sent occasionally.

jagoyette, 2018-03-31:
I stumbled across this problem also and submitted a pull request that resolves the issue (see https://github.com/cesanta/mongoose/pull/900). The fix uses MG_PEEK to increment the receive buffer size until the whole datagram can be read.

cpq, 2020-12-14:
I guess a note in the documentation about the compile-time limit would do the job.

cpq, 2020-12-17:
Closing this.

Added note and a link to this issue to https://cesanta.com/docs/#build-options

closed 2020-12-17 by cpq