cesanta/mongoose #909: mg_find_index_file supporting virtual directories on ESP32 SPIFFS

avanbremen, 2018-03-30:
I ran the Mongoose [simplest web server](https://github.com/cesanta/mongoose/blob/master/examples/simplest_web_server/simplest_web_server.c) example project on ESP32 using [Mongoose 6.11](https://github.com/cesanta/mongoose/releases/tag/6.11) with mg_broadcast [fix](https://github.com/cesanta/mongoose/issues/899), [ESP-IDF Pre-release 3.0-rc1](https://github.com/espressif/esp-idf/releases/tag/v3.0-rc1) and [SPIFFS file system](http://esp-idf.readthedocs.io/en/latest/api-reference/storage/spiffs.html).

**Feature request**
Because SPIFFS does not support directories, an index.html is not automatically served by the web server, nor is `s_http_server_opts.enable_directory_listing = "yes"` producing any output. From a technical standpoint this behaviour is however expected, as mongoose.c checks `if (is_directory)` before calling e.g. `mg_find_index_file`. Therefore this is more of a feature request, then an issue.

When e.g. mounting SPIFFS file system under /spiffs, setting `s_http_server_opts.document_root = "/spiffs/web/"` and placing an index.html in /spiffs/web/, I think it would be a great feature if index.html is automatically served and directory listing works.

**The issue**
/spiffs/web/index.html is not actually placed under directory /spiffs/web, rather a file is created in SPIFFS called /web/index.html.

> Presently, spiffs does not support directories. It produces a flat structure. If SPIFFS is mounted under /spiffs creating a file with path /spiffs/tmp/myfile.txt will create a file called /tmp/myfile.txt in SPIFFS, instead of myfile.txt under directory /spiffs/tmp.

**Proposed change**
Perhaps an addition could be made so that the path is checked for path separators, which results in virtual (sub)directories being detected, index.html being served and dir listing working. Perhaps this behaviour could be depending on a config define.

Thanks again for creating on Mongoose!

snahmad, 2018-11-27:
Hi,

How about SD card. I mount SD card on ESP32. I specify that root folder for mongoose web server. Can I browser folders?

cpq, 2020-12-17:
That functionality is something that does not belong to the mongoose core.
Instead, this could be implemented as a custom esp32-specific event handler.

closed 2020-12-17 by cpq